Botspot/wor-flasher #89: Failed to download uupdump.zip, exiting now.

FlarosOverfield, 2022-07-13:
## Setup:

- Which version of Windows you chose: Windows 11
- Raspberry Pi model: 3B+
- Name of SD card you tried to flash: Toshiba 32GB
- What day you first downloaded wor-flasher: 13/07/2022
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc): Raspberry Pi OS 64Bit

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
Error 500: Internal Server Error.
I heavily recommend not using uupdump anymore because it's a dumpster housefire.

ksaro1, 2022-07-15:
I get same error 500 from uupdump.net at the stage that the Windows ISO file is being downloaded.

Botspot, 2022-07-15:
Going to this link in a web browser returns a more descriptive error.
[image]

I plan to make a mitigation for this soon.

ksaro1, 2022-07-15:
> Going to this link in a web browser returns a more descriptive error. [image]
> 
> I plan to make a mitigation for this soon.

That's interesting. Because my previously succesful wor-flasher install now fails to boot with an error that Windows file is missing, which is why I was re-installing.

Botspot, 2022-07-15:
This was harder to implement than expected, but I eventually came to a satisfactory result with commit https://github.com/Botspot/wor-flasher/commit/da06adfb00132f975d3643d3dc3fbe5f44ae007b.

closed 2022-07-15 by Botspot