Botspot/wor-flasher #90: Error codes

tannerdennis, 2022-07-17:
## Setup:

- Which version of Windows you chose- **Windows 11**
- Raspberry Pi model- **Pi4**
- Name of SD card you tried to flash (some are too slow or too small)- **Kingston 240GB SSD**
- What day you first downloaded wor-flasher- **7/17/2022**
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)- **Debian**

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

**https://www.youtube.com/watch?v=WqFr56oohCE** 

I followed this video step by step.
I am having issues I am flashing my SSD. I keep having the following errors pull up while they are downloading.

07/17 08:17:52 [ERROR] CUID#429 - Download aborted. URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/651f52f3-aa2d-4c12-8fae-71d8ab4b18ff?P1=1658066790&P2=404&P3=2&P4=HOVePQBit0zhk%2bwgp%2fPtbKCoFqaSkhOy39BAEsVYEaD7fVgo6oRGFeaFPA0Pe0I5VeXLYd6GxyctQ9vEnTNKHg%3d%3d
Exception: [AbstractCommand.cc:351] errorCode=1 URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/651f52f3-aa2d-4c12-8fae-71d8ab4b18ff?P1=1658066790&P2=404&P3=2&P4=HOVePQBit0zhk%2bwgp%2fPtbKCoFqaSkhOy39BAEsVYEaD7fVgo6oRGFeaFPA0Pe0I5VeXLYd6GxyctQ9vEnTNKHg%3d%3d
  -> [SocketCore.cc:1018] errorCode=1 SSL/TLS handshake failure: Error in the pull function.

07/17 08:17:52 [ERROR] CUID#426 - Download aborted. URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/651f52f3-aa2d-4c12-8fae-71d8ab4b18ff?P1=1658066790&P2=404&P3=2&P4=HOVePQBit0zhk%2bwgp%2fPtbKCoFqaSkhOy39BAEsVYEaD7fVgo6oRGFeaFPA0Pe0I5VeXLYd6GxyctQ9vEnTNKHg%3d%3d
Exception: [AbstractCommand.cc:351] errorCode=1 URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/651f52f3-aa2d-4c12-8fae-71d8ab4b18ff?P1=1658066790&P2=404&P3=2&P4=HOVePQBit0zhk%2bwgp%2fPtbKCoFqaSkhOy39BAEsVYEaD7fVgo6oRGFeaFPA0Pe0I5VeXLYd6GxyctQ9vEnTNKHg%3d%3d
  -> [SocketCore.cc:1018] errorCode=1 SSL/TLS handshake failure: Error in the pull function.

07/17 08:17:52 [ERROR] CUID#427 - Download aborted. URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/651f52f3-aa2d-4c12-8fae-71d8ab4b18ff?P1=1658066790&P2=404&P3=2&P4=HOVePQBit0zhk%2bwgp%2fPtbKCoFqaSkhOy39BAEsVYEaD7fVgo6oRGFeaFPA0Pe0I5VeXLYd6GxyctQ9vEnTNKHg%3d%3d
Exception: [AbstractCommand.cc:351] errorCode=1 URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/651f52f3-aa2d-4c12-8fae-71d8ab4b18ff?P1=1658066790&P2=404&P3=2&P4=HOVePQBit0zhk%2bwgp%2fPtbKCoFqaSkhOy39BAEsVYEaD7fVgo6oRGFeaFPA0Pe0I5VeXLYd6GxyctQ9vEnTNKHg%3d%3d
  -> [SocketCore.cc:1018] errorCode=1 SSL/TLS handshake failure: Error in the pull function.
[DL:195KiB][#5e29a7 44MiB/66MiB(67%)][#49acae 19MiB/68MiB(28%)][#72c4b7 7.7MiB/81MiB(9%)][#d8be77 32KiB/83MiB(0%)][#ae826e 1.5MiB/166MiB(0%
07/17 08:17:59 [ERROR] CUID#447 - Download aborted. URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/62922772-0d5f-4fd1-915b-e78c77a1f81b?P1=1658065130&P2=404&P3=2&P4=ac7FderZWKcx5uRvhfnDgJGm7MpvrnfGMXHRwAB5vDjsGXtiMHsxev%2f%2bG9uEoac6nuCWyxK8iCi7GNw2MgpK7Q%3d%3d
Exception: [AbstractCommand.cc:351] errorCode=1 URI=https://uupdump.sf.tlu.dl.delivery.mp.microsoft.com/filestreamingservice/files/62922772-0d5f-4fd1-915b-e78c77a1f81b?P1=1658065130&P2=404&P3=2&P4=ac7FderZWKcx5uRvhfnDgJGm7MpvrnfGMXHRwAB5vDjsGXtiMHsxev%2f%2bG9uEoac6nuCWyxK8iCi7GNw2MgpK7Q%3d%3d
  -> [SocketCore.cc:1018] errorCode=1 SSL/TLS handshake failure: Error in the pull function.

Botspot, 2022-07-17:
The trickiest part of the entire flashing process is the necessity to generate an ARM64 Windows image. Microsoft distributes ready-made Windows ISOs for x86 computers, but for some reason not for ARM. So WoR-Flasher has to use https://uupdump.net to generate an image from hundreds of separate files on Microsoft servers.

If your internet connection is unreliable at all, you will likely encounter connection errors. Uupdump's script tries to restart failed URLs, and in addition to that, WoR-Flasher retries the uupdump script four times before giving up.
@tannerdennis, did WoR-Flasher give up and say the uupdump script failed, or did you cancel the process when you saw the errors?

tannerdennis, 2022-07-17:
I did not cancel anything. When I came back, this error had popped up.

An error has occurred:
Failed to copy /home/pi/wor-flasher-files/isomount/sources/install.wim to /media/pi/WOR-installer/winpart
Exiting now.

Botspot, 2022-07-17:
> I did not cancel anything. When I came back, this error had popped up.
> 
> An error has occurred: Failed to copy /home/pi/wor-flasher-files/isomount/sources/install.wim to /media/pi/WOR-installer/winpart Exiting now. @Botspot

That's an entirely different error. Interesting. Was there any notable output before the red error message? I would imaging that if install.wim was missing, the `cp` command would say there was "no such file or directory".

tannerdennis, 2022-07-17:
> > I did not cancel anything. When I came back, this error had popped up.
> > An error has occurred: Failed to copy /home/pi/wor-flasher-files/isomount/sources/install.wim to /media/pi/WOR-installer/winpart Exiting now. @Botspot
> 
> That's an entirely different error. Interesting. Was there any notable output before the red error message? (Look in the terminal, not the pop-up dialog) I would imagine that if install.wim was missing, the `cp` command would say there was "no such file or directory".

This was at the bottom of the terminal, 
cp: error writing '/media/pi/WOR-installer/winpart/install.wim': Input/output error
cp: failed to close '/media/pi/WOR-installer/winpart/install.wim': Input/output error
Failed to copy /home/pi/wor-flasher-files/isomount/sources/install.wim to /media/pi/WOR-installer/winpart

I also found this in the middle, 
UUP Converter v0.7.0
Note: This script does not and cannot support the integration of updates.
Use the Windows version of the converter to integrate updates.

Botspot, 2022-07-17:
> This was at the bottom of the terminal,
>
> ```
> cp: error writing '/media/pi/WOR-installer/winpart/install.wim': Input/output error
> cp: failed to close '/media/pi/WOR-installer/winpart/install.wim': [...]

Interesting. That would indicate a hardware failure on your end. This has nothing to do with your Internet at all! Perhaps the storage device became unresponsive partway through. Try flashing again.

> I also found this in the middle,
> 
> ```
> UUP Converter v0.7.0
> Note: This script does not and cannot support the integration of updates.
> Use the Windows version of the converter to integrate updates.
> ```

That's normal. On the https://worproject.com website there are instructions for updating Windows manually.

tannerdennis, 2022-07-18:
@Botspot 
It completed the download successfully! However, I am trying to reboot the system and it keeps showing this. Any ideas?
[image]

tannerdennis, 2022-07-18:
Not sure if that photo will work. It says, “couldn’t find the installation disk” 
My ssd is plugged in

Botspot, 2022-07-18:
Could you send a picture of your physical setup and of the screen? For the image to come through, wait for it to finish uploading before clicking the "Comment" button.

Botspot, 2022-07-18:
Also, the “couldn’t find the installation disk” error has been encountered by others before.
See: https://github.com/Botspot/wor-flasher/issues/34
I've never been able to reproduce this particular error, but on the issue I linked above, [someone said](https://github.com/Botspot/wor-flasher/issues/34#issuecomment-922468920) they discovered it was due to their particular storage hardware failing to fallback to a 512 byte sector size.

tannerdennis, 2022-07-18:
[image]
[image]

Botspot, 2022-07-20:
Try using a different SSD adapter and/or SSD. Due to the sector-size issue which I explained above, not all hardware works out-of-the-box.
If you have a large enough SD card on hand, you can also try installing to that.

tannerdennis, 2022-07-30:
@Botspot update
I was able to use a SD card to get windows working. However, I am stuck on this screen because none of the usb ports are working. 
[image]

tannerdennis, 2022-07-30:
So, I can plug in this bluetooth speaker to charge it through the usb ports and it will work. It does not want to work with a wireless or wired keyboard and mouse.

tannerdennis, 2022-07-30:
[image]
I tried restarting the pi to see if that would help. The mouse works until the loading bar shows up on this screen.

tannerdennis, 2022-08-06:
@Botspot any ideas?

Botspot, 2022-08-06:
@tannerdennis, could you try updating your Pi4's bootloader? This is easily done by running `sudo rpi-eeprom-update -a` in Raspberry Pi OS.

Also, please try re-flashing Windows with WoR-Flasher, as [I recently fixed a bug](https://github.com/Botspot/wor-flasher/issues/92) that was causing people to get outdated versions of Windows.

tannerdennis, 2022-08-06:
@Botspot I updated the bootloader and re-flashed windows. It is doing the same thing

tannerdennis, 2022-08-17:
@Botspot any other ideas to fix my issue?

Botspot, 2022-08-28:
> @Botspot I updated the bootloader and re-flashed windows. It is doing the same thing

If a windows image has already been downloaded, then WoR-Flasher uses that and avoids re-generating it. Please try deleting the ~/wor-flasher-files folder and re-flash.

closed 2024-02-10 by Botspot