Botspot/wor-flasher #91: Does wor-flasher supports Windows 8/8.1?

ijamon, 2022-07-30:
Hello! Just a question in title.

Purpose - build an rpi for good sounding with a special scipt which turn's of not-need functionality of Windows.

Botspot, 2022-07-30:
> Hello! Just a question in title.

I'm not sure if Windows 8 ever had a public ARM release, but even if it did at some point, Microsoft would probably not be hosting any updates for it anymore. Furthermore, I've never been able to find any mention of "Windows 8" on uupdump.net - the place that coordinates the creation of Windows ISO generation.

> Purpose - build an rpi for good sounding with a special scipt which turn's of not-need functionality of Windows.

That really depends on what functionality you need. It varies per person. Do you want OneDrive? Do you want an antivirus? What about all the other drivers and DLLs?
Anyway, this processes of simplifying Windows my removing stuff is called **debloating**. There are many tutorials on the Web for how to do this.
One place that I think yields good perspective on debloating was a few concurrent replies on the Raspberry Pi Forum thread about this wor-flasher tool. https://forums.raspberrypi.com/viewtopic.php?p=1970214#p1969903

ijamon, 2022-07-31:
Thanks! I already found something for this.

closed 2022-07-31 by ijamon