Botspot/wor-flasher #92: Kmode Exception Not Handled BSOD after flashing

MEMESCOEP, 2022-07-30:
## Setup:

- Which version of Windows you chose 
Windows 11

- Raspberry Pi model
Raspberry Pi Model B 4GB Model

- Name of SD card you tried to flash (some are too slow or too small)
PNY Elite-X 64gb

- What day you first downloaded wor-flasher
Today

- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
Raspberry Pi OS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
All I did was flash Windows 11 onto the SD Card by following this tutorial: https://www.maketecheasier.com/install-windows-raspberry-pi/#install-windows

wrssmithjr04, 2022-08-01:
Same here this is my setup: 
 
## Setup:

- Which version of Windows you chose
- Windows 11 Insider Preview 25169.1000 (rs_prerelease) 

- Raspberry Pi model
- Raspberry Pi Model B 4GB Model

- Name of SD card you tried to flash (some are too slow or too small)
- Sandisk Ultra 128 GB USB 3 

- What day you first downloaded wor-flasher
- 8/1/2022

- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
- Raspberry Pi OS 64 bit

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
I cloned the wor flasher repo to my home directory then ran the non gui version of it.
I also used custom version providing the update id of ee9cee18-d029-4263-a7a8-e2ccbc05382a

wrssmithjr04, 2022-08-02:
After testing for a bit i have found out that build 22621.317 of windows 11 works and boots but i have not tested the insider versions. 
The update id for the build is `2f9e556e-193b-4875-bb43-74516c4e5a82` while running the script you will have to select custom when selecting the version and enter this id

closed 2022-08-02 by Botspot

Botspot, 2022-08-02:
I have just solved a portion of this problem. More details below.

> * What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
>   Windows 11 x86_64

@MEMESCOEP, I'm not sure how you could be running WoR-Flasher on Windows 11 x86_64, as WoR-Flasher is a Linux-only tool and is mainly used on Raspberry Pi OS.

> I also used custom version providing the update id of `ee9cee18-d029-4263-a7a8-e2ccbc05382a` The drive was also made using the recovery drive option

That is currently the most recent (unstable) Windows Insider build, and I'm not too surprised you found it unstable.

> After testing for a bit i have found out that build 22621.317 of windows 11 works and boots but i have not tested the insider versions.
> The update id for the build is [...]

This is interesting, @wrssmithjr04. That one simple finding of yours helped me identify an oversight in wor-flasher.
Due to download failures from uupdump in the past, I recently changed wor-flasher to check if the given update-id is downloadable. **If it is not,** wor-flasher will try the next Update ID on the list. What list? It's a list of Update-IDs derived from a search for "windows 11 22h2 arm64".
Anyway, WoR-Flasher would try the first one: `39d7cde6-94bf-4dca-a7c2-ed0abc0970f3`, and sure enough it would fail with "error 500 internal server error". This is a problem with uupdump.
Next, wor-flasher would try `2f9e556e-193b-4875-bb43-74516c4e5a82`, and it would also fail. The `wget` tool would exit with an error code. But this time, the server was actually returning a *different* code. This time it said "429 Too Many Requests", but wor-flasher treated it as if the whole Update ID was bad, and it would move on to the next one. Eventually, after enough time passes, it will succeed, but by then several update-ids have been skipped.

In other words, WoR-Flasher has been several Windows 11 releases behind because it misinterpreted uupdmp server return codes. I've addressed this in commit https://github.com/Botspot/wor-flasher/commit/8b89bda05e18cc1b76162fb3175c82549bf6a61a, so now WoR-Flasher should choose `2f9e556e-193b-4875-bb43-74516c4e5a82`, until another stable Windows version is released.

I'm going to assume the "Kmode exception not handled" was just an instability in the Windows Insider builds. If that assumption is incorrect, please let me know and I can try to flash a card with windows to see what can be done. This issue was closed for now, as I fixed WoR-Flasher's tendency to pick an unnecessarily old Windows version.

MEMESCOEP, 2022-08-02:
@Botspot Yup, sorry I was running it on raspberry pi os. I accidentally put Windows 11 x86_64 because I was using something from the wor website. I'll see if the fix works, but before I use it I'll say that I did try it on raspberry pi os and I got the same bsod with a boot loop. Cheers!

wrssmithjr04, 2022-08-02:
What build are you using? The latest insiders of win 11 won't work you have to go with the 22H2 version. @MEMESCOEP

MEMESCOEP, 2022-08-02:
@wrssmithjr04 Ah, I see. I'm using the latest build lol. I'll test if the 22h2 build will work

wrssmithjr04, 2022-08-02:
If you use the example id that I said worked above you should be good. @MEMESCOEP

MEMESCOEP, 2022-08-02:
@wrssmithjr04 how do you choose a different build version?

wrssmithjr04, 2022-08-02:
For the Linux version you will press 3.) custom when you run it in the terminal and use `2f9e556e-193b-4875-bb43-74516c4e5a82` as the id.

For the windows version you will need to go to [this link](https://uupdump.net/selectlang.php?id=2f9e556e-193b-4875-bb43-74516c4e5a82) in order to download the iso 
@MEMESCOEP

MEMESCOEP, 2022-08-02:
@wrssmithjr04 alr. I was using the gui on rpi os lol. I'll try it later as soon as I get off work

Botspot, 2022-08-02:
> If you use the example id that I said worked above you should be good. @MEMESCOEP

And as an extension, if you simply select "Windows 11" from the tool, it will also pick a suitable up-to-date ID. No need to provide it manually, when WoR-Flasher would use the same one anyway.

commit 8b89bda0 (2022-08-02): handle uupdump rate-limiting to fix #92

Without this, many newer Windows releases were being skipped because
they all 'failed', when in reality it was a server-side rate-limiting
mechanism.