Botspot/wor-flasher #94: ISO Mount Error

Lolcat2112, 2022-08-25:
## Setup:

- Which version of Windows you chose
Windows 10
- Raspberry Pi model
Raspberry Pi 4
- Name of SD card you tried to flash (some are too slow or too small)
USB Flash Drive, like mentioned in this video: https://youtu.be/nc-6G9R2APE
- What day you first downloaded wor-flasher
Today
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
Raspberry Pi OS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

Followed The Instructions like in said video above, but while it was installing, The ISO Couldn't Mount... I've sent a picture of what it looks like on my side.
[image]

Botspot, 2022-08-26:
The important error here is "module udf not found in directory /lib/modules...".

Udf is a kernel module that allows ISO images to be mounted to the filesystem. For some reason, loading `udf` failed on your system.

This means one of two things:

- You upgraded your kernel and have not rebooted yet. This causes the `modprobe` command to continue looking in the old location for modules, when in fact they have been moved to a new folder for the new kernel version.
- You are running an abnormal Linux kernel that does not have the `udf` module.

Lolcat2112, 2022-08-26:
Ok. I'll try rebooting. Updated the kernel yesterday, just before starting the install for WoR-flasher

Lolcat2112, 2022-08-26:
It worked! Thank you!

[image]

closed 2022-08-26 by Lolcat2112

Botspot, 2022-08-27:
@Lolcat2112, I have updated wor-flasher to display a more specific error if this happens again.

See: https://github.com/Botspot/wor-flasher/commit/01fbe073133bc62c0dbaca5a55ddffb8cc81a86a

commit 01fbe073 (2022-08-27): Use specific errors if kernel modules are missing to close #94