Botspot/wor-flasher #96: Random error  on the format of the USB key

fredmo, 2022-09-09:
Maybe it could be successful to inform you : 

Random error on the format of the USB key :  tested with 2 different sticks. ( I am on ubuntu 22.04 ) 

Stick 1 : 
- GPT 
umount: bad usage
Try 'umount --help' for more information.
Creating partition table
Error: Partition(s) on /dev/sdb are being used.
Failed to make GPT partition table on /dev/sdb!

Stick 2 : 
- mkfs vfat error , missing device
but on the third retry, it was functional. 

Regards,

fredmo, 2022-09-09:
another try, another log : 

umount: bad usage
Try 'umount --help' for more information.
Creating partition table
Generating partitions
Generating filesystems
Partition 1: /dev/sdc1, Partition 2: /dev/sdc2
mkfs.fat 4.2 (2021-01-31)
exfatprogs version : 1.1.3
Creating exFAT filesystem(/dev/sdc2, cluster size=32768)

Writing volume boot record: done
Writing backup volume boot record: done
Fat table creation: done
Allocation bitmap creation: done
Upcase table creation: done
Writing root directory entry: done
Synchronizing...

exFAT format complete!
Mounting /dev/sdc device to /media/xxx/WOR-installer
FUSE exfat 1.3.0+git20220115
fuse: mountpoint is not empty
fuse: if you are sure this is safe, use the 'nonempty' mount option
Failed to mount /dev/sdc2. Trying again after loading the 'fuse' kernel module.
FUSE exfat 1.3.0+git20220115
fuse: mountpoint is not empty
fuse: if you are sure this is safe, use the 'nonempty' mount option
Failed to mount /dev/sdc2 to /media/xxx/WOR-installer/winpart

MillowBro, 2023-08-22:
Damn, I went across this same problem rn, how do I fix it?

closed 2024-02-10 by Botspot