Botspot/wor-flasher #98: uup_download_linux.sh failed, most likely due to unreliable Internet. Trying again in 1 minute. (Attempt 4 of 4) Failed to generate a Windows ISO! uup_download_linux.sh exited with an error so please see the errors above."

unksoldr, 2022-09-17:
## Setup:
Windows 11
Pi 4 8gb
Sandisk Ultra 32GB
13SEP22
Mint

First time I flashed it tried to boot,  made it to the Windows Setup screen but no further. I rewrote the SD card and got the error.  Used 2 different older updates, but they give the same error.

The old posts about this are a year old and obviously it has not been solved.

Botspot, 2022-09-17:
> uup_download_linux.sh failed, most likely due to unreliable Internet. Trying again in 1 minute. (Attempt 4 of 4) Failed to generate a Windows ISO! uup_download_linux.sh exited with an error so please see the errors above."

Please post the errors.

mrmrmystery, 2022-09-17:
I have the same Error

Error:
```
Downloading converters...
Exception caught
Exception: [AbstractOptionHandler.cc:69] errorCode=28 We encountered a problem while processing the option '--checksum'.
  -> [OptionHandlerImpl.cc:395] errorCode=1 Unrecognized checksum

We have encountered an error while downloading files.

uup_download_linux.sh failed, most likely due to unreliable Internet.
Trying again in 1 minute. (Attempt 4 of 4)
Failed to generate a Windows ISO! uup_download_linux.sh exited with an error so please see the errors above.
```

Botspot, 2022-09-17:
If multiple people encountered this error, then it's probably not an "unreliable internet" issue. As this "Unrecognized checksum" error seems to be occurring within the uup_download_linux.sh script.

I have reached out to a UUPdump maintainer about this issue. Now we wait... :)

awuctl, 2022-09-18:
> If multiple people encountered this error, then it's probably not an "unreliable internet" issue. As this "Unrecognized checksum" error seems to be occurring within the uup_download_linux.sh script, it is most likely an issue on uupdump's end.
> 
> I have reached out to a UUPdump maintainer about this issue. Now we wait... :)

maybe reach out in an issue next time instead of to the wrong maintainer

The first checksum in `files/converter_multi` file is wrong.
You can remove it for now, we'll fix it in a bit.

eraseyourknees, 2022-09-18:
uup-dump-dev/website@8154c8c98cd7e738533d9946b3a5e56938a0915c

Botspot, 2024-02-10:
This has been fixed

closed 2024-02-10 by Botspot