Botspot/wor-flasher #99: Installation disk needs to be recreated in order to try again

on4xb, 2022-09-23:
At the preinstall evironment i get an error and the installation fails
Error
Installation disk needs to be recreated in order to try again
Hardware is a raspberry 3b+ with 1 tb usb 3.1 external SSD. At this point the 64GbSDcard has been removed as per instruction.
I assume the the build of the ISO was succesfull

Any pointers/help pse

Bert

Botspot, 2022-09-24:
First, as a warning, I don't maintain the Preinstallation Environment. So if something goes wrong there, there are few things I can do, other than to show the error to a WoR maintainer, who usually has no idea how to help. :(

Could you take a picture of your screen and attach it to this Github Issue?

on4xb, 2022-09-24:
raspberry 3b+ with 1 tb usb 3.1 external SSD
[image]
[image]

on4xb, 2022-09-28:
Hello,
Today (and after a long wait) i received my PI4-8Gb. Itried again to install Windows 11 and went like a charm. I used the same SDcard and SSD. In other words the error happens only with the rpi3B+ version. Tnx for the support anyway

Bert

Botspot, 2022-10-14:
> After several failed attempts I found out that my micro SD card was the problem. At the time, I was using a Samsung Evo card which never worked and today I found another SD card around the house called Samsung Evo Ultra which worked.
> 
> Most Samsung card's are really good but I bought this messed up one really really super cheap on Amazon. I am thinking it was probably a fake card from a 3rd party seller. The card that worked was bought at best buy

Perhaps Wor-Flasher can be improved to detect cards like yours. Could you insert the defective card into your pi with a usb adapter, then send me the output of `lsusb`?

susanbles, 2023-01-15:
I have got this problem too, installing windows 10 on rpi4 4Gb. I tried to instal from buster 32 bit on sandisk 64Gb flashdrive and from 64bit bullseye on 32 Gb class10 micro Sd  on sandisk flashdrive  and from buster 64 bit on sandisk 64Gb flashdrive on forementioned micro Sd. All used cards an drives work well with other OS's.

Botspot, 2023-01-16:
> I have got this problem too, installing windows 10 on rpi4 4Gb. I tried to instal from buster 32 bit on sandisk 64Gb flashdrive and from 64bit bullseye on 32 Gb class10 micro Sd on sandisk [...]

Flash drives are not recommended and SD cards are hit or miss.
Could you click the "Logs" button and send a picture of the error?

susanbles, 2023-01-25:
Thanks for your advice, another micro sd card did the trick.

closed 2023-01-25 by Botspot