Botspot/wor-flasher #103: cant select dl dir the example dont work give an example so i can dl to a ext4 partition on a usb will you

jamieduk, 2022-10-04:
cant select dl dir the example dont work give an example so i can dl to a ext4 partition on a usb will you

jamieduk, 2022-10-04:
Ok heres how i managed to do it despite bs insttructions and crappy examples

1st format usb drive as ext4

then mount

mount -t ext4 /dev/sbd1 /mnt/

then edit the gui .sh file and find dl_dir and put in following (line 28)
DL_DIR="/mnt/wor-flasher-files"

then save and run the sh file you just edited!

simple!!!

jamieduk, 2022-10-04:
Ok so far so good heres how i dont it this is real instructions not the bs the uploader tried to write!!

git clone https://github.com/Botspot/wor-flasher.git

gui based:

To run WoR-flasher using the graphical interface

Ok heres how I managed to do it despite bs instructions and crappy examples

1st format usb drive as ext4

then mount

mount -t ext4 /dev/sbd1 /mnt/

then edit the gui .sh file and find dl_dir and put in following (line 28)
DL_DIR="/mnt/wor-flasher-files"

then save and run the sh file you just edited!

simple!!!

make sure installation files are moved to a location with space more than 16gb

now make sure you own the folder and files 

sudo chown jay wor-flasher/*

then make executable

sudo chmod +x *.sh

~/wor-flasher/install-wor-gui.sh

After install has been successful run raspberrypi-imager and select boot options and install usb bootloader on a usb, not the 1 you just put setup files on but another 1 then reboot!

Botspot, 2022-10-06:
> then edit the gui .sh file and find dl_dir and put in following (line 28)
> DL_DIR="/mnt/wor-flasher-files"
> 
> then save and run the sh file you just edited!

While this works fine, did it really fail to to use DL_DIR when you tried it the recommended way?
This is the recommended way:
```
DL_DIR=/mnt/wor-flasher-files ~/wor-flasher/install-wor-gui.sh
```
Also, there is a GUI way to change the DL_DIR. On the final dialog of WoR-Flasher, there will be an "Advanced..." button that lets you set the "Working directory".

> now make sure you own the folder and files
> 
> then make executable
> 
> sudo chmod +x *.sh

Was this really necessary for you? The `git clone` command should make the scripts executable by default, because when I uploaded these files to the wor-flasher git repository, I made sure they were executable on my end.

> ~/wor-flasher/install-wor-gui.sh
> 
> After install has been successful run raspberrypi-imager and select boot options and install usb bootloader on a usb, not the 1 you just put setup files on but another 1 then reboot!

This is unnecessary. All you need to do is to have a recent bootloader (which should be automatically supplied by `apt upgrade`), and USB boot works fine, out of the box, if no SD card is found.
You can even go into `sudo raspi-config` to change the Boot Order. The Pi can try booting from USB first, before it tries to boot from SD card, but by default it will try SD first and then USB.

closed 2022-10-06 by Botspot