Botspot/wor-flasher #105: Stuck on the black screen with big rpi-logo and a endless running circle

pengbins, 2022-11-04:
## Setup:

- Which version of Windows you chose

 Selected Windows version: Windows 11 build 22621.755

- Raspberry Pi model

RPI_MODEL: 4

- Name of SD card you tried to flash (some are too slow or too small)

sumsung ssd 850 evo (250G)  + sataToUSB3 cabel

- What day you first downloaded wor-flasher

2022-Nov-04

- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)

Ubuntu 20.04.5 LTS  amd64

## Procedure:
1. run ./install-wor.sh    with no error info
2. unplug ssd from wor-flasher machine, plug to rpi4b, and boot rpi4b
3.  Stuck on the black screen with big rpi-logo and a endless running circle

pengbins, 2022-11-04:
But wor.exe on Windows works.

closed 2023-02-24 by pengbins

Botspot, 2023-02-24:
Did you get it resolved, @pengbins?

pengbins, 2023-04-28:
> Did you get it resolved, @pengbins?

Not with `wor-flasher` on Ubuntu, but with `wor.exe` on Windows works. So I didn't try newer versions anymore.