Botspot/wor-flasher #107: cant install (is this a issue with uup dump?)

hiidiotthisisakshatpasbola, 2022-12-13:
well over the last week nothing seems to work 
the website(uup dump is down i guess)
but i thought to inform devs 
so yes its screwed up.

Botspot, 2022-12-13:
Yes, uupdump seems to be undergoing a DDOS attack right now. :(
I am on the uupdump discord server band someone there said it has been like this since Saturday. I hope they fix it soon!

Vexcited, 2022-12-23:
Hello, I am getting an error:

```console
get_uuid(): Failed to get list of Windows 11 versions.
Please check if <a href="https://uupdump.net">uupdump.net</a> can be reached from your web browser.
```

Is it related to this issue? It seems like I can reach `uupdump.net` from my web browser without any issue but the script is still complaining.

Botspot, 2022-12-23:
Yes, this is related. While the uupdump site is (mostly) back up, their API domain is still down.
You can verify this yourself by going to https://api.uupdump.net

I am currently working on a revamped WoR-Flasher that is faster, more reliable, and does not rely on uupdump. I expect to be done by today or tomorrow.

Botspot, 2022-12-24:
Okay everyone, I have just uploaded my changes to this github repository. Please try it again and let me know how it goes.

pradeeps, 2022-12-24:
Thank you @Botspot . It went past the error and works now.

closed 2022-12-24 by Botspot