Botspot/wor-flasher #108: Wor flasher not posting

dogehy, 2022-12-21:


dogehy, 2022-12-21:
[image]

Botspot, 2022-12-21:
Turn off your pi, and then turn it on again. This time please don't press any keys and in 30 minutes or so it should reach a purple screen that will begin the installation process.
Lack of a network will display a warning on this screen, but it should not prevent the system from reaching the installation screen. I think you just need to give it more time.

dogehy, 2022-12-21:
thanks so much for replying. however, 3 hours later and nothing has happened. is there a problem with the version of windows i'm using? are there any known problems with windows 10?![Uploading 209002583...][image]

Botspot, 2022-12-22:
Are you using a Pi3 or a Pi4?
What is the brand and model of your sd card?

dogehy, 2022-12-22:
pi four 4gb, pny 64gb card

Botspot, 2022-12-22:
> pi four 4gb, pny 64gb card

What is the speed rating on that sd card? A1? Class 10?
Do you have any other SD cards or usb drives that you can try?

closed 2024-02-10 by Botspot