Botspot/wor-flasher #109: System.ComponentModel.Win32Exception (14): Not enough memory resources are available to complete this operation

charles-gray, 2022-12-28:
## Setup:

- Windows 10 Pro
- Raspberry Pi 3B
- Failure installing to SanDisk 64GB Ultra USB drive
- Using wor-flasher git tag '59d5a6f'
- wor-flasher ran cleanly on 64b raspberry pi OS and device booted to installer

## Procedure:

1) Made a "Create a recovery drive to install Windows on other >16 GB drives" SD card (16GB) of Windows 10 en-us
2) Booted raspberry pi 3b to a 1GB UEFI SD device in the on-board slot. It automagically detected the installer SD (attached via adapter to USB).
3) Ran installer on a 64GB SanDisk ultra USB thumb drive also attached to USB
4) Part way through `Applying the WIM image to: H`, I get an out of memory error.

I've tried this a few different ways with a few different combinations (16gb root disk, install to 32gb SD card), and always get the exact same error at about the same time. See attached log. I tried messing with video settings to get more RAM but that didn't help. Any suggestions appreciated

[WoR-PE-log.log](https://github.com/Botspot/wor-flasher/files/10311606/WoR-PE-log.log)

Botspot, 2022-12-28:
Thanks for reaching out. Also I commend your effort to troubleshoot this and I appreciate details you've provided. It makes this a lot easier.
As I actually didn't write the Preinstallation Environment, I am not sure what options you have at this stage. The actual WoR-Flasher seems to worked correctly in this case.

Right now I can say with a fairly high confidence that the reason you are encountering this error is because you are using a Pi with 1GB of RAM.

I've sent this issue to the developer of the Preinstallation Environment. He may have some ideas to make this step succeed.

In the meantime, try adding this line to config.txt of the sd device:
```
gpu_mem=64
```
I think 64 is the minimum value.

charles-gray, 2022-12-28:
Thanks for the prompt reply! Apologies for the misfiled bug report. The WoR-Flasher does indeed seem to be working just fine.

I tried the `gpu_mem=64` suggestion, but sadly got the same result.

Is there a bug tracker I can follow along for the Preinstallation Environment? I'm still learning my way around the Windows install environment.

Botspot, 2022-12-28:
> Is there a bug tracker I can follow along for the Preinstallation Environment? I'm still learning my way around the Windows install environment.

Well the WinPE environment itself is a simple OS developed by microsoft. But the WoR installer program which _runs inside_ this environment, I have been calling it the Preinstallation Environment. In other words this is not a problem with Microsoft's software.
I wish the developer for this tool would host releases and source code on github, but right now he distributes it with Google Drive. http://worproject.com/dldserv/worpe/downloadlatest.php

> Thanks for the prompt reply! Apologies for the misfiled bug report. The WoR-Flasher does indeed seem to be working just fine.

I do not consider this misfiled because there is no better place to ask for help. The only other place I can think of would be on a WoR community chat. https://worproject.com/community

> I tried the `gpu_mem=64` suggestion, but sadly got the same result.

Try `gpu_mem=32`
Also I think this could be fixed by actually shrinking the `install.wim` filesize. This can be accomplished by removing the `--soft` flag from line 955 of install-wor.sh. The line in its entirety should look like this:
```bash
  errors="$(wimdelete "$SOURCE_FILE" 1 2>&1)" || error "Failed to remove a partition from $SOURCE_FILE\nErrors:\n$errors" #remove --soft for this last one to minimize filesize
```
Once you do this please delete the `winfiles*` directory in your `wor-flasher-files` (Or DL_DIR) folder to force a rebuild of the Windows image.

Botspot, 2022-12-28:
I was informed by the developer of the PE installer that the memory requirements are increasingly tighter with each new iteration of Windows.
He theorizes that the reason why you are encountering ram-related issues now, whereas it used to succeed on Pi3s a year ago, could be the increased size of Windows. So it may be worthwhile to try a year- or two year old version of Windows 10.

charles-gray, 2022-12-29:
Thanks! I tried the other other suggestions (removing --soft, gpu_mem=32) but to no avail. I'll try and wind back the clock on Windows. I've not tried an older version, but I assume someone must have?

charles-gray, 2022-12-29:
No such luck, sadly. I wound back to `17134.112 (2018-07-05)`. It failed to start the spinner. Moving forward to `18362.356 (2019-09-12)`, it started the installer but only far enough to complain I needed version `19041` for compatibility. So I tried `19041.508 (2020-10-19)` and still got the same out of memory error.

Googling around, I also found gpu_mem=16 might be valid, but that never got into EFI. Trying to dial back the display parameters in EFI (with a higher gpu_mem parameter), it always seems to use the full resolution of the panel. I'll contemplate if I can find a low-res panel and dial back the memory then.

Any extra suggestions welcome in the mean time!

Botspot, 2022-12-29:
> Thanks! I tried the other other suggestions (removing --soft, gpu_mem=32) but to no avail. I'll try and wind back the clock on Windows. I've not tried an older version, but I assume someone must have?

Well I recall that this used to be possible on a Pi3. Since it worked in the past but failed now, potentially the difference could be the version (and primarily size) of the latest Windows.

closed 2024-02-10 by Botspot