Botspot/wor-flasher #110: Wifi

dogehy, 2023-01-09:
## Setup:

- Which version of Windows you chose
- Raspberry Pi model
- Name of SD card you tried to flash (some are too slow or too small)
- What day you first downloaded wor-flasher
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
[image]

When I try to use windows 10, the wifi won't work. I have a TP-Link adapter if that has any use. Is there any known fixes?

Botspot, 2023-01-09:
See the readme.
[image]

Botspot, 2023-01-09:
There is one working driver for a particular wifi usb dongle. I don't know the details of it. You can consult https://worproject.com - I think the info is available somewhere - maybe in their discord.

closed 2023-01-09 by Botspot

Botspot, 2023-01-27:
Ethernet works though.