Botspot/wor-flasher #111: Wifi icon not show on windows 11

asrafnasri, 2023-01-27:
## Setup:

- Which version of Windows you chose (windows 11 pro)
- Raspberry Pi model (4b 8gb)
- Name of SD card you tried to flash (some are too slow or too small) m.2 ssd 150gb samsung
- What day you first downloaded wor-flasher (january 2023)
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS)

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

When windows has installed wifi icon not shown and no wifi settings in windows setting, i didn't try do anything yet

Itai-Nelken, 2023-01-27:
As far as I know there are no drivers for the onboard rpi wifi chip. 
The ethernet port works and you can use hotspot over bluetooth as well I think.

Botspot, 2023-01-27:
@Itai-Nelken is correct.

Please refer to the main information found at https://github.com/Botspot/wor-flasher.
[image]

closed 2023-01-27 by Botspot