Botspot/wor-flasher #112: Windows 10 On RPI3 B+ Not Booting

GoldenEagle75, 2023-01-28:
## Setup:

- Which version of Windows you chose
Windows 10
- Raspberry Pi model
RPI3 B+
- Name of SD card you tried to flash (some are too slow or too small)
Kingston 32GB Micro SDHC
- What day you first downloaded wor-flasher
Today (1/28/23)
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
Raspberry Pi 64 bit

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

I attempted to install Windows 10 on my RPI3 from a different RPI3. I had no problem at all installing the software and then flashing it. After the SD card was ejected, I plugged it into my RPI and there was nothing. The HDMI monitor displayed its "no signal" screen and the green light on the RPI board did not blink or turn on. The red light was solid the entire time.

Is this because of the "dtoverlay=upstream-pi4" setting on config.txt?

symbios24, 2023-05-26:
Hello im not an expert on this but when i tried on my pi3 in an sd card it could not complete the installation,better try on an ssd or a fast flash drive a memory card is not fast enough for this in my opinion

axelromfelt, 2023-05-28:
> Hello im not an expert on this but when i tried on my pi3 in an sd card it could not complete the installation,better try on an ssd or a fast flash drive a memory card is not fast enough for this in my opinion

Hello, did you get it to work on a flashdrive? if so how did you get it to boot from the flash drive

Botspot, 2023-05-29:
> Hello, did you get it to work on a flashdrive? if so how did you get it to boot from the flash drive

You should just be able to flash the flash drive, shutdown the pi, unplug any sd card but leave the flash drive inserted, then the Pi will boot from the flash drive.
If that does not work, update the **Bootloader** of your Pi. This is easy to do with a google search.

Botspot, 2024-02-10:
Hello again. I think the issue has been that the Pi4 firmware was being put on the sd card instead of the pi3 firmware. This has now been fixed. Could you try it again?

GoldenEagle75, 2024-02-18:
Hello,
I have again tried and continue to have the same problem.
I'll try again soon on a different raspberry pi and notify you if this fixes

commandcontrolQ, 2024-08-09:
I am experiencing the same issue.

I am using a Raspberry Pi 3B and when trying to install Windows 10, the microSD card will not boot **at all**. The screen does not update, and the light on the Pi remains red.

I am certain this is some sort of regression because **I have gotten Windows 10 to work on this Pi model before**, so I know this is possible.

This could be because I used the Windows version, however the difference in platform should not cause any issues.

I will make changes to the config.txt and notify about any differences.

commandcontrolQ, 2024-08-27:
### I have found a fix.
It appears that **running the flasher on Windows** fixes the boot issue.

This is likely due to the fact that the Windows flasher _skips the pre-install environment_ by **installing Windows to the SD card directly.**

For some reason, there must be some sort of issue that prevents Windows PE from starting properly on the Raspberry Pi 3B (and 3B+).

TheJosemi98, 2024-09-04:
Hello!

I want to confirm that the last comment has helped me to run Windows 10 in a Rpi4 Model B. The red and green led remained solid in all booting attempts of the wor-flasher image.

Once I flasher the SD card by the Windows flasher version, the boot issue disappears