Botspot/wor-flasher #113: Packages not installing on Arch Linux

ShaunTheVyonder2008, 2023-01-31:
## Setup:

- Which version of Windows you chose
Windows 11
- Raspberry Pi model
-4B
- Name of SD card you tried to flash (some are too slow or too small)
ScanDisk 16GB
- What day you first downloaded wor-flasher
Just now
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
Arch Linux

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

I'm installing Windows 11 ARM with the tool, but however it failed due to the packages not corrected, and because my OS in different to ubuntu, debian, etc.

I'm not sure if these packages are working for Arch for me right now.

Can someone help me out?

Botspot, 2023-02-01:
Well here is the list of debian packages that need installing.
```
yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse wget udftools
```
Maybe they all have direct 1-to-1 equivalents on Arch? I don't know.

Try running this command:
```
sudo pacman -S yad aria2 cabextract wimtools chntpw genisoimage exfat-fuse wget udftools
```
Please report back if that command succeeds on your system. If it does, I can see to it that WoR-Flasher skips trying to install the deb versions of those tools.

ShaunTheVyonder2008, 2023-02-03:
I just got it working with a different script that someone in the community made, and now works.

Thanks anyway!

closed 2023-02-03 by ShaunTheVyonder2008

Botspot, 2023-02-03:
Well if arch users keep using Worli, then I suppose there is no reason to add Arch support to WoR-Flasher!

In any case, it is good to know you got it working. Enjoy Windows. If that is even possible. ;)