Botspot/wor-flasher #114: Unable to select Tiny 11 iso file. File size is too small (kinda the point)

xWass, 2023-02-15:
NOTE: I AM ATTEMPTING TO INSTALL A NON STANDARD VERSION OF WINDOWS 11 BASED ON Windows 11 Pro 22H2. THIS IS FOR TESTING

Is there some way to bypass the under 3gb iso restriction?

## Setup:

- Which version of Windows you chose: Tiny 11 (https://www.minitool.com/news/tiny11-download-iso.html)
- Raspberry Pi model: 4b
- Name of SD card you tried to flash: sandisk 32gb
- What day you first downloaded: 3 days prior to this
- What linux operating system you are using to run wor-flasher on: rasppi os

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
select OS, error under 3gb iso

xWass, 2023-02-15:
oops

xWass, 2023-02-15:
Update: I was able to remove the file size check in the terminal (not gui) script.

closed 2023-02-15 by xWass

Botspot, 2023-02-16:
> NOTE: I AM ATTEMPTING TO INSTALL A NON STANDARD VERSION OF WINDOWS 11 BASED ON Windows 11 Pro 22H2. THIS IS FOR TESTING
> 
> * Which version of Windows you chose: Tiny 11 (https://www.minitool.com/news/tiny11-download-iso.html)

Unless you found an **ARM** version of Tiny11, there is no way it could work on a Pi. What most people mean whan they say "Windows 11" is the Intel/AMD version. But WoR-Flasher has to generate the ARM version of Windows in order for a Pi to boot it, because the Pi's CPU is an ARM chip.

Botspot, 2023-02-20:
Here is an announcement from the primary leader of the Windows on Raspberry project:

> Important announcement
> 
> Recently there has been a huge influx of Tiny 11 videos floating around the internet and people trying it out.
Since it now comes in an ARM64 flavor i am required to make this announcement
> 
> Tiny 11 is not supported by this project
> 
> If you wish to use Tiny 11 please use it at your own risk
> If you experience any bugs or issues while using it, Do not ping staff for support.
> If you want a light windows experience you may use the provided NTlite scripts in 📦-downloads as they have been verified to not break core system components while keeping the system debloated.
> 
> Please note that Tiny 11 is also against Microsoft's ToS
> This server used to provide premade debloated images in the past but due to us wanting to keep things
compliant with microsoft's ToS we decided to stop doing that.
> Microsoft does not really care about people redistributing windows images, However it is still against
the laws so please keep that in mind.