Botspot/wor-flasher #115: Error while starting installation of Windows 10 on RPi 400

MrBurgerCZ, 2023-02-22:
## Setup:

- Which version of Windows you chose
Windows 10
- Raspberry Pi model
RPi 400 4GB
- Name of SD card you tried to flash (some are too slow or too small)
USB 3.1 32GB Flash Drive
- What day you first downloaded wor-flasher
20. February 2023
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
Twister OS
## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

I downloaded WoR Flasher from Pi-Apps
Launched WoR Flasher
Selected Win 10 on RPi 400
Selected the flash drive
I left the default settings
I started the installation
No errors during installation
I booted from the flash drive
And I see this error
[image]

Botspot, 2023-02-22:
Thank you for taking the time to ask for help! I am glad to help you.

> * Name of SD card you tried to flash (some are too slow or too small)
>   USB 3.1 32GB Flash Drive

That is your problem. Windows is surprisingly picky when it comes to boot drives. If yours is not fast enough, windows will completely refuse to boot from it.

Most other people who have reported this error here have had good success with a SSD or a fast A1-rated SD card.

closed 2023-02-22 by Botspot

MrBurgerCZ, 2023-02-22:
That's weird. With this flash drive I had a working Win 11. Thanks for the answer.

MrBurgerCZ, 2023-02-22:
@Botspot

Botspot, 2023-02-22:
Others have reported that Windows 10 fails to install on drives that Windows 11 works on. I have no control over it - this is all locked behind closed doors with Microsoft.

See: https://github.com/Botspot/wor-flasher/issues/34
And https://github.com/Botspot/wor-flasher/issues/93

MrBurgerCZ, 2023-02-22:
Thank you. I'll try to install it on SD card running Twister OS.