Botspot/wor-flasher #117: Couldn't find the installation disk!

eksapetm, 2023-03-02:
I use wor-flasher on my laptop (Ubuntu GNU/Linux 22.04) to install Windows 10 (en-us) for a Raspberry Pi 4 (8 GB). The installation process works fine, but when I try to boot it a window pops up with the following error:

```
Couldn't find the installation disk!
Make sure it is connected to your device and has all the files necessary, then click on the reload button.
If the error persists, please check the log for more details.
```
By now I have tried to use wor-flasher on many different storage devices: an external 240GB SSD, several USB thumb drives, but also from an SD card. Also I have tried to boot from an USB thumb drive and leave the external SSD attached (to install it on the SSD), but it always returns the same error.

Unfortunatel,y, I can't click on OK because I do not have a USB-mouse. I do have a USB-keyboard attached, but it does not respond to any key press.

I am not sure whether my problem is caused by wor-flasher, but since this is the tool that I used to flash I figured to first post it here.

Any help is appreciated. Thanks!

Botspot, 2023-03-02:
Try updating your Pi's bootloader.
https://jamesachambers.com/raspberry-pi-4-bootloader-firmware-updating-recovery-guide/

eksapetm, 2023-03-04:
> Try updating your Pi's bootloader. https://jamesachambers.com/raspberry-pi-4-bootloader-firmware-updating-recovery-guide/

Thanks for your response. I realize I should have mentioned this, but I already did that. I am up-to-date on the default channel:

```
$ sudo rpi-eeprom-update 
BOOTLOADER: up to date
   CURRENT: Wed Jan 11 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed Jan 11 17:40:52 UTC 2023 (1673458852)
   RELEASE: default (/lib/firmware/raspberrypi/bootloader/default)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

... but changing this to beta shows nu updates:

```
$ sudo rpi-eeprom-update -a
BOOTLOADER: up to date
   CURRENT: Wed Jan 11 17:40:52 UTC 2023 (1673458852)
    LATEST: Wed Jan  4 10:27:49 UTC 2023 (1672828069)
   RELEASE: beta (/lib/firmware/raspberrypi/bootloader/beta)
            Use raspi-config to change the release.

  VL805_FW: Using bootloader EEPROM
     VL805: up to date
   CURRENT: 000138c0
    LATEST: 000138c0
```

This is probably because [the previous beta became stable in january 2023](https://github.com/raspberrypi/rpi-eeprom/blob/master/firmware/release-notes.md#2023-01-12---promote-previous-beta-release-to-stable).

OceaneML, 2023-04-01:
Hello, I have the same problem to boot windows 10 on USB with SSD key.
I do some test :
-With Windows 11 no problem (the SSD boot well on USB)
-With Windows 10 and SD card no problem
Why Windows 10 won't work with SSD on USB ???
I also need help. Thanks

Fabiano2510, 2023-06-06:
Hi, I have the same problem. What I have to do?!

WB8AZP, 2023-08-25:
This still gives me the "Couldn't find the installation disk!" message.  I am using a 32Gig Samsung EVO drive, I have updated my Raspberry Pi 4 B EEPROM log file says:
Exit code: -2147024809
Couldn't mount the partitions
Couldn't find PE disk!

jimmythesnitch, 2023-08-29:
Same here, Tried from my PC (Linux Mint)  as well as from Raspberry Pi OS 32gig SD to a 128samsung flash drive.

"Couldn't find the installation disk!"

Test Mode says  "Build 19041.vb.release.191206-1406

No USB keyboard or mouse works on the WoR-PE

Works in UEFI settrings configs etc.

updated bootloader to:
   
CURRENT: Thu 11 May 06:26:03 UTC 2023 (1683786363)
    LATEST: Thu 11 May 06:26:03 UTC 2023 (1683786363)

maciek8844, 2023-08-31:
i have same problem, but i today i fix it. 
Do this: 
https://jamesachambers.com/raspberry-pi-4-bootloader-firmware-updating-recovery-guide/

"sudo nano /etc/default/rpi-eeprom-update

Change the line FIRMWARE_RELEASE_STATUS=”critical” to:

FIRMWARE_RELEASE_STATUS="stable" "

and update from 11 May, 

Next ,very important run script wor-flasher without GUI !
~/wor-flasher/install-wor.sh

Botspot, 2023-08-31:
@maciek8844, good tip on changing the firmware release status. I have not had this problem, but it is possible that newer UEFI versions need a newer bootloader.

Why exactly is it bad to run wor-flasher with the GUI? What happened when you did that?

maciek8844, 2023-08-31:
I don't know what was bad in GUI version, i try it over 5 hours and same error "Couldn't find the installation disk!" , with SD Card, with SSD, with Pendrive. 
I try run script in terminal ~/wor-flasher/install-wor.sh , and now almost finish installtion windows

Botspot, 2023-08-31:
Well the GUI version will run the install-wor.sh script in the background, so I am not exactly sure what could be different when running install-wor.sh yourself.
Did you update the bootloader before trying both GUI and CLI versions of wor-flasher?

maciek8844, 2023-08-31:
Yes

maciek8844, 2023-08-31:
U have one more option, in terminal you can change where you want to install windows, you can to choose install to this same disk or another disk over 16GB.

Nova1545, 2023-10-31:
I have run into the exact same problem, anyone have any further ideas for how to fix this issue?
Using a PI4 with the latest firmware installed (Followed the suggestions farther up in the issue before adding my comment here, nothing)

closed 2024-02-10 by Botspot