Botspot/wor-flasher #118: Very long boot time

HQv2, 2023-03-13:
## Setup:

- **Which version of Windows you chose:** 11
- **Raspberry Pi model:** 4B 8GB
- **Name of SD card you tried to flash (some are too slow or too small):** Sandisk 128GB
- **What day you first downloaded wor-flasher:** 2023-03-13
- **What linux operating system you are using to run wor-flasher on:** Raspberry Pi OS

Hello,

Windows 11 has installed successfully and works well, but every time I power on or reboot the device, it hangs on the black Raspberry window for about 10 minutes. Is this normal?

[image]

Thank you!

HQv2, 2023-03-14:
Resolved by changing the boot order. It then started getting stuck on automatic repair and wouldn't boot, so I gave up on this.

closed 2023-03-14 by HQv2

Botspot, 2023-03-17:
The 10 minute delay happens on every boot for everybody. From what I understand, this is due to some issue on Window's side. As nobody but Microsoft understands how Windows works, the free world is more or less powerless to improve it.

Your options are:
- To stop using Windows (recommended)
- To put up with the delay
- To hack microsoft and leak the Windows source code so some WoR developers can make workarounds. (obviously, not recommended at all)