Botspot/wor-flasher #121: Wor will not install

jfrommer2001, 2023-04-02:
Lib modules  kernel is 6.1.19-v7+,  6.1.19+, 6.1.19-v7l+ 6.1.19-v8+. and wor won't install.
It is looking for 5.4.81-rt45-v8+
Please advice
Tks
Jack

Botspot, 2023-04-03:
Hello @jfrommer2001, please send me the full terminal output.

If you already closed it, then try rebooting your Pi. Usually when "wrong kernel version" errors occur, rebooting works because it means you updated the kernel and have not yet rebooted.

Botspot, 2023-04-05:
@jfrommer2001, have you been able to solve this issue yet?

jfrommer2001, 2023-04-07:
No
It does request 5.xx.xxx  library
please advise

jfrommer2001, 2023-04-07:
Error: " The running kernel  (5.4.81-rt45-v8+)  does not match any
directory in /lib/modules ......"
Rebooting did not solve the error
Please advice

Botspot, 2023-04-10:
Hello @jfrommer2001, could you tell me more about your linux system you are running wor-flasher on?

I am not exactly sure what the problem is here, but your reported kernel version alone (5.4.81-rt45-v8+) indicates to me that your system is not the average Raspberry Pi OS install. I want WoR-Flasher to work in your case, but before I can do that, I need to understand what your system is and where it stores its kernel modules if they are missing in the normal /lib/modules location.

Please run these commands and report back what they say. (they are not intended to fix your system, they will help me understand how your kernel is set up)
```
modprobe --show-depends fuse
modprobe --show-depends udf
```

closed 2024-02-10 by Botspot