Botspot/wor-flasher #125: [ERROR] CUID#8 - Exception caught while allocating file space.

karl1poiss, 2023-05-02:
## Setup:

- Which version of Windows you chose - Windows 10
- Raspberry Pi model - Raspberry Pi 400
- Name of SD card you tried to flash (some are too slow or too small) - Toshiba 16 GB
- What day you first downloaded wor-flasher - Within the last 2 weeks
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) - Raspberry Pi Os

## Procedure:
1. Windows 10 on a Raspberry Pi 4/400
2. en-us
3. dev/sda/ 14.5 GB Toshiba TransMemory
4. Do nothing to the config.txt. Just flash.
During the fisrt steps of flashing these errors pop up:
[ERROR] CUID#8 - Exception caught while allocating file space.
Exception: [AbstractDiskWriter.cc:557] errNum=28 errorCode=9 fallocate failed. cause: No space left on device
[ERROR] CUID#8 - Download not complete: /home/karlk/wor-flasher-files/winfiles_17134.112_en-us/image.esd
Does anyone now how to fix this issue?

Botspot, 2023-05-03:
You have two storage devices that BOTH need enough space. The first device is the one your Pi booted from and will store all the Windows files that are downloaded. The second device will actually be flashed with Windows. Both devices need sufficient space.

In your case, it appears that a download failed because your currently booted device did not have enough space. To see what is going on, please run two commands for me and report back what they say:
```
rm -rf ~/wor-flasher-files
df -B 1 ~/
```
Explanation: I want to see how much free storage space you had just before running wor-flasher.
- The `rm` command removes your partially downloaded windows files, essentially returning your free space to how it was before. This command should not return any output.
- The `df` command will display how many bytes of storage memory are free in your booted storage drive.

closed 2023-05-28 by karl1poiss

Botspot, 2023-05-29:
@karl1poiss, were you able to solve the problem you were having?

karl1poiss, 2023-06-01:
Yes, I did but now i am having another issue.