Botspot/wor-flasher #128: Kernel not updating to /Lib/modules

StarFireBoss, 2023-05-20:
### System
I am running linux debian 10 on a Chromebook (arm64)
### Issue
I cloned the repository and ran the GUI script but I got this:

DIRECTORY: /home/starfireboss/wor-flasher
DL_DIR: /home/starfireboss/wor-flasher-files
lsblk: /dev/vdc[/lxd/storage-pools/default/containers/penguin/rootfs]: not a block device
Checking for internet connection... Done
The running kernel (5.15.108-18908-gd6447a7dca06) does not match any directory in /lib/modules.
Usually this means you have not yet rebooted since upgrading the kernel.
Try rebooting.
If this error persists, contact Botspot - the WoR-flasher developer.
xkbcommon: ERROR: Key "<CAPS>" added to modifier map for multiple modifiers; Using Mod3, ignoring Lock

Any info would be helpful
Thanks,

Botspot, 2023-05-21:
Since you closed this issue, I will assume that you followed the suggestion in the error message (reboot), and that it solved the problem for you.

StarFireBoss, 2023-05-21:
sorry I closed it by accident and could not find the page afterward.  Same problem but I tried rebooting and updating around 5 times now.

Botspot, 2023-05-22:
> sorry I closed it by accident and could not find the page afterward. Same problem but I tried rebooting and updating around 5 times now.

Please run these commands in a terminal and reply back with the output from each of them.
```
uname -r
ls /lib/modules
```

Botspot, 2023-05-22:
I just now noticed that you wrote that you are using a chromebook. My guess would be that your chromebook does not support standard kernel module loading. It may be possible through a non-standard way, which I would be happy to support, or it may not be possible.

StarFireBoss, 2023-05-22:
uname -r:
5.15.108-18908-gd6447a7dca06
ls /lib/modules:
ls: cannot access '/lib/modules': No such file or directory

Botspot, 2023-05-22:
Try running these commands:
```
sudo modprobe udf && echo succeeded
sudo modprobe fuse && echo succeeded
```
See if both commands result in a message that says "succeeded".

Botspot, 2023-05-24:
@StarFireBoss, were you able to try the latest commands I asked you to run?

StarFireBoss, 2023-05-24:
modprobe: FATAL: Module udf not found in directory /lib/modules/5.15.108-18910-gab0e1cb584e1
modprobe: FATAL: Module fuse not found in directory /lib/modules/5.15.108-18910-gab0e1cb584e1

Botspot, 2023-05-25:
Optional extensions to your linux kernel are stored in /lib/modules. The fact that your entire /lib/modules folder does not exist means one of two things:
1. Your system is improperly configured and you need to reinstall the kernel.
2. Your system is working as intended and chromebooks use some minimized or unusual kernel setup when running linux distros.

I am not sure which possibility is more likely. You can try researching the solution yourself, or you can try two other approaches:
1. If you have a Windows computer nearby, you can flash WoR to your Raspberry Pi SD card using that. Instructions at https://worproject.com
2. Since you must already have a Raspberry Pi to install windows on, you could run Raspberry Pi OS on it first, and use wor-flasher on the Pi to flash a secondary SD card.

I am closing this issue, not because I want to end the conversation here but because I do not think I can be of any further help with this exact problem you have reported. Feel free to reply back with which approach you chose to take, and any further questions.

closed 2023-05-25 by Botspot