Botspot/wor-flasher #129: Cursor dosent show up in the PE

karl1poiss, 2023-05-29:
## Setup:

- Which version of Windows you chose -  Windows 10
- Raspberry Pi model - Raspberry Pi 400
- Name of SD card you tried to flash (some are too slow or too small) - Toshiba 16 GB
- What day you first downloaded wor-flasher - Dont remember
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) - Raspberry Pi OS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
- Flash the device
- Go to Device Manager and under there Raspberry Pi Settings
- Set resolution to 720p and turn off native resolution
- Turn off 3 GB ram limit
- Save and boot the PE

karl1poiss, 2023-06-01:
Also, this happened when I decided to reinstall the WOR-Flasher because I thought that the PE was not installing the OS (it was, and very very very slowly.)

lolimjustherefornoreason, 2024-02-14:
Same here when I was in the installer

Botspot, 2024-02-14:
Try updating the bootloader.

lolimjustherefornoreason, 2024-02-17:
I'm a beginner at this stuff so I don't understand how to update the bootloader

lolimjustherefornoreason, 2024-02-17:
And when I booted to Installer it was showing to plug in a storage to install to even though I was using a 64 GB Usb stick which can install Windows on itself

Argetlam3, 2024-03-11:
Having this issue as well. My mouse isn’t showing up, I’m going to troubleshoot and will see what happens

WombleWoo7547, 2024-04-14:
If anyone is still here, to update the bootloader, boot into **Raspberry Pi OS** (lite or desktop). In a terminal, type in:
```bash
rpi-eeprom-update
```
and follow the instructions.

karl1poiss, 2025-01-16:
I have also noticed that no method of input works, not just the mouse. Even the integrated keyboard doesn't work.

karl1poiss, 2025-01-17:
Fixed this issue by inserting a SD card when the machine booted up, causing it to install Windows even though the mouse dosen't work.