Botspot/wor-flasher #131: not installed

Ivomir1, 2023-08-05:
PLS can you help me with wor-flasher? I install on raspbian and started. Flashed my USB SSD drive with windows 11. Every works fine. Windows downloaded, flashed. Now i have this window. When i press shutdown my raspberry 4 turned off and nothing else. Windows 11 has not installed. When i press ON my raspberry turned on in old raspbian system.

[image]

Botspot, 2023-08-05:
You need to make your Pi boot from the USB SSD. In your case, it sounds like you just need to remove your sd card to force the Pi to boot from the SSD instead of the SD card.

Botspot, 2023-08-07:
Any updates, @Ivomir1?

closed 2024-02-10 by Botspot