Botspot/wor-flasher #134: Worflasher won't install windows 11, it says failed to make GPT partition table (on raspberry pi)

MillowBro, 2023-08-22:
## Setup:

- Windows 11
- Raspberry Pi model 4 2GB
- Lexar USB 8GB
- The time I'm posting this I installed Wor-Flasher
- I'm using Raspberry Pi OS

## Procedure:
All I did is went to worflasher, clicked options, clicked choose specific windows version, then I chose the top one, then I continued. And it popped up, Before I did had problems with the USB tho, I currently don't have any other available, I think it's because of the USB, With partitions inside or smt

Botspot, 2023-08-22:
Does the error occur on the purple screen that you boot to, or was this an error you saw in the black terminal window while still running wor-flasher from Linux?

Botspot, 2023-08-22:
Also, it is a bad idea to try to boot from any flash drive. Flash storage like that will wear out quickly if you do it.
SD cards are better, and they are especially important for as slow an OS as Windows. I recommend you flash your Pi's SD card with Windows. If you only have a USB drive and a sd card, try flashing Raspberry Pi OS (Linux) to the flash drive, then boot from the flash drive and flash the sd card with windows.

Botspot, 2023-08-22:
Closing the issue now, as it is 99% probably an issue with your specific flash drive you tried to flash. This is not something I can fix from a software side. Feel free to reply to this issue, even after it is closed.
Good luck! :)

closed 2023-08-22 by Botspot

MillowBro, 2023-08-23:
Ok, back, It was the usb :P, I fixed it with a few simple cmds on my windows pc

Botspot, 2023-08-23:
> Ok, back, It was the usb :P, I fixed it with a few simple cmds on my windows pc

Let me guess... you used diskpart to format it.