Botspot/wor-flasher #135: No boot & WIM not valid

zieglerjm, 2023-08-25:
I have a RPi 3b.  The downloaded WIM for Windows 10 & 11 on 8/25/2023 is not a valid WIM for DISM to extract?  

- Assuming this should be a valid WIM for a Windows PC?

Also, I can't get the image to boot from SD card or USB SSD?  

- Is there a way to update the boot loader or turn on debugging?

Botspot, 2023-08-26:
What was the extraction errors?

zieglerjm, 2023-08-27:
Sorry, I didn't include more details.  Is there a way to get a direct link vs using the Linux tool?

I can list the contents just fine: 
- dism /Get-ImageInfo /imagefile:{wim}
Export gets the below error on all 3 versions of 19045.2965_en-us & 22621.1702_en-us: 
- DISM /Mount-image /imagefile:{wim} /Index:1 /MountDir:{folder}
  - Error: 11
  - An attempt was made to load a program with an incorrect format.

C:\Windows\Logs\DISM\dism.log
`2023-08-26 20:49:23, Info                  DISM   DISM.EXE: Dism.exe version: 10.0.22621.1
2023-08-26 20:49:23, Info                  DISM   DISM.EXE: Executing command line: DISM  /Mount-image /imagefile:{path}\install.wim /Index:{1/2/3} /MountDir:{path}
2023-08-26 20:49:23, Info                  DISM   DISM Imaging Provider: PID=624 TID=7916 WIM image specified - CGenericImagingManager::GetImageInfoCollection
[624.7916] [0x8144012d] 
2023-08-26 20:49:23, Warning               DISM   DISM WIM Provider: PID=624 [ReadWimHeader:(1906) -> header version mismatch] {wim} (HRESULT=0x8007000B) - CWimManager::WimProviderMsgLogCallback
[624.7916] [0xc144012e] 
2023-08-26 20:49:23, Error                 DISM   DISM WIM Provider: PID=624 [WIMCreateFile:(419) -> Fail to read WIM header] {wim} (HRESULT=0x8007000B) - CWimManager::WimProviderMsgLogCallback
2023-08-26 20:49:23, Error                 DISM   DISM WIM Provider: PID=624 TID=7916 "WIM open failed." - CWimImageInfo::Mount(hr:0x8007000b)
2023-08-26 20:49:23, Error                 DISM   DISM Imaging Provider: PID=624 TID=7916 onecore\base\ntsetup\opktools\dism\providers\imagingprovider\dll\genericimagingmanager.cpp:833 - CGenericImagingManager::InternalOpMount(hr:0x8007000b)
2023-08-26 20:49:23, Error                 DISM   DISM Imaging Provider: PID=624 TID=7916 onecore\base\ntsetup\opktools\dism\providers\imagingprovider\dll\genericimagingmanager.cpp:746 - CGenericImagingManager::InternalCmdMount(hr:0x8007000b)
2023-08-26 20:49:23, Error                 DISM   DISM Imaging Provider: PID=624 TID=7916 onecore\base\ntsetup\opktools\dism\providers\imagingprovider\dll\genericimagingmanager.cpp:538 - CGenericImagingManager::ExecuteCmdLine(hr:0x8007000b)
2023-08-26 20:49:23, Info                  DISM   DISM.EXE: Image session has been closed. Reboot required=no.
`

Botspot, 2024-02-10:
Pi3s could not boot because this was mistakenly downloading pi4 uefi files only. This is now fixed. Try it again.

closed 2024-02-10 by Botspot