Botspot/wor-flasher #136: Successfully downloaded ESD image, but it appears to be corrupted. Please run this script again.

saumyajit, 2023-09-11:
## Setup:

- Which version of Windows you chose : Windows 11
- Raspberry Pi model : RPI4B 8GB
- Name of SD card you tried to flash (some are too slow or too small) : 128 GB SSD
- What day you first downloaded wor-flasher : 11 September 2023
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) : Raspberry Pi OS

## Procedure:

Input configuration:
DL_DIR: /home/testuser1/wor-flasher-files
RUN_MODE: gui
RPI_MODEL: 4
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 0
BID: 22621.1702
WIN_LANG: en-us
CONFIG_TXT: ⤵
  > # don't change anything below this point #
  > arm_64bit=1
  > arm_boost=1
  > enable_uart=1
  > uart_2ndstage=1
  > enable_gic=1
  > armstub=RPI_EFI.fd
  > disable_commandline_tags=1
  > disable_overscan=1
  > device_tree_address=0x1f0000
  > device_tree_end=0x200000
  > dtoverlay=miniuart-bt
  > dtoverlay=upstream-pi4
CONFIG_TXT: ⤴
DRY_RUN: 1

Downloading WoR PE-based installer from Google Drive
Done ———————————————————————————————————————————————————————————————————————————

Downloading ARM64 drivers
Done ———————————————————————————————————————————————————————————————————————————

Downloading Pi4 UEFI firmware
Done ———————————————————————————————————————————————————————————————————————————

Downloading Windows ESD image
Done ———————————————————————————————————————————————————————————————————————————
Verifying download... Done

Successfully downloaded ESD image, but it appears to be corrupted. Please run this script again.
Expected SHA1 hash is 5d429a7fb2cd9d3dad4636a2b09d3c8d587f69dd, but downloaded file has SHA1 hash 046a0e580903d5357b66e35dff78a3cee6055310

3 times I'ved tried to download & install which even not happened. please suggest

Botspot, 2023-09-11:
First, thank you for posting the full terminal output. This helps a lot, and it's not something everybody does. So thank you for doing that.

The SHA1 hash check is very important. It makes sure that the file you downloaded exactly matches the file on the server. If there is a mismatch, then the download could have been interrupted, corrupted, or in extreme cases, even intercepted by malicious hackers. The integrity check helps avoid downstream issues caused by a corrupted download at this step.

I have a few questions:
1. How stable is your internet connection? Are you using WiFi or Ethernet?
2. You may notice that in the last line of output, it tells you the expected SHA1 hash and the actual SHA1 hash. Is the actual SHA1 hash always the same for you, or has it been something different every time you retried?

I think that this could be one of three things:
1. This issue is caused by you having an unreliable internet connection
2. The server is giving an incorrect SHA1 hash (extremely unlikely)
3. You partially downloaded the ESD image the first time you ran wor-flasher, and since then wor-flasher has been trying to use that when it is not supposed to. [This happened recently, but I thought I fixed it.
](https://github.com/Botspot/wor-flasher/issues/132)

To help rule out possibility number 3, does the green progress bar for "Downloading Windows ESD image" take a few minutes, or does it finish from 0% to 100% in a second or two? (It is a multi-gigabyte file - it should take a few minutes)

saumyajit, 2023-09-12:
`How stable is your internet connection? Are you using WiFi or Ethernet?` - I'm using Wifi & it's quite stable
`You may notice that in the last line of output, it tells you the expected SHA1 hash and the actual SHA1 hash. Is the actual SHA1 hash always the same for you, or has it been something different every time you retried?` -- no, actual SHA1 hash for all 3 esd downloads was different every time. 

`To help rule out possibility number 3, does the green progress bar for "Downloading Windows ESD image" take a few minutes, or does it finish from 0% to 100% in a second or two? (It is a multi-gigabyte file - it should take a few minutes)` -- the ESD file download took almost 45 - 60 minutes to complete on all 3 occasions.

please do let me know for anymore details which I can provide.

Botspot, 2023-09-12:
Thank you for the quick reply. I will make a suggestion for your future replies on github in general: it's easier to quote somebody else's words by making each line of the quotation start with `> `. Github even does this automatically for you if you click "Quote reply" inside the 3 dots on a message you want to reply to.

@saumyajit, your answers have confirmed that my possibility #2 is the case here.
Particularly this part confirms it:

> no, actual SHA1 hash for all 3 esd downloads was different every time.

It would be stranger if the SHA1 hash was the same each time, but because it's different, it shows that your network connection is unstable. I think that if you ran `ping google.com` in a terminal for an hour and then pressed Ctrl+C to cancel it, you would see some failure rate.
I would recommend moving closer to the wifi router, removing nearby metal objects, and avoiding using the internet for other things. This ESD image is a big file, and even if one byte of that file is incorrectly downloaded, it's no good. Keep trying until it works. I cannot help you further.

At this point I will be closing the issue, because I feel that there is nothing more I can do to help you. Feel free to still reply to this issue if you have more to say about this issue.

closed 2023-09-12 by Botspot