Botspot/wor-flasher #138: is it possible to install win 11 to pi 5?

gapito, 2023-11-10:
## Setup:

- Which version of Windows you chose
- Raspberry Pi model 
- Name of SD card you tried to flash (some are too slow or too small)
- What day you first downloaded wor-flasher
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

taellinglin, 2024-02-11:
it's currently installing windows on my pi-5. I had to take the SD card out at boot and put it back in for it to start the installer for some reason or else it would just boot to a blank screen...

Botspot, 2024-02-11:
@anvme @kyle0122 @nrck @gapito @gabriel-visualthinking wor-flasher now supports pi5
Closing the issue as completed.

closed 2024-02-11 by Botspot