Botspot/wor-flasher #140: Failed to create FAT partition on /dev/sdb1

aghosh0605, 2023-12-31:
## Setup:

**- Which version of Windows you chose**
   Running on Ubuntu 20.04LTS
- **Raspberry Pi model**
   Raspberry Pi 4 Model B
- **Name of SD card you tried to flash (some are too slow or too small)**
   Sandisk 128GB High Speed Micro SD Card
- **What day you first downloaded wor-flasher**
   Today
- **What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)**
   Debian

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

1. After running the tool with `./install-wor-gui.sh` and setting it to _Windows 10_ and all default settings on the parent machine where I am creating the flash drive.
2. After the format process completes some errors are coming. Below are the screenshots
[image]
[image]

**I tried with the Windows tool also. It is also stuck after the format process and after assigning some of the disk partition values like V, G, S etc. As Windows shows no errors and the logs of the tool also showed no error I am unable to post any screenshots of Windows but I can feel some problem is happening inside after opening the task manager as after a certain time the disk usage of the removable media goes to zero**

Botspot, 2024-01-02:
It looks like the sd card was removed during installation. This was likely accidental, and I have some USB adapters that are sensitive to being bumped or nudged.
Did you try this several times and always encounter the same error? What are you using to connect the sd card to your system? Is it a usb adapter or are you usb booting and writing to the sd card directly inserted into the pi's built-in sd card slot?
Low voltage powering the pi can cause peripheral disconnects. Do you see any low voltage warnings on the screen? See if this reports anything: `vcgencmd get_throttled`

aghosh0605, 2024-01-02:
> It looks like the sd card was removed during installation. This was likely accidental, and I have some USB adapters that are sensitive to being bumped or nudged. Did you try this several times and [...]

I was using a dongle with an SD card slot, and maybe it is slow or has some issues. I wasted my whole night just for a USB dongle 😂  Then I bought a new SD card reader and now it works amazingly. Thanks for the help 😊

Botspot, 2024-01-02:
Awesome! That's determination.
Good to know it's solved. I will close the issue now. Feel free to open a new one if you encounter any other difficulties.

Enjoy Windows (if that's even possible lol)

closed 2024-01-02 by Botspot