Botspot/wor-flasher #141: Error applying the WIN image to B:

sayuso99, 2024-01-08:
## Setup:

- I tried with Windows 10 and Windows 11, same error for both
- Raspberry Pi 4 (4GB)
- SD Kingston (32GB)
- 20 of december of 2023
- Raspberry Pi OS (32bit)

## Procedure:
I used wor-flasher in a Raspberry Pi OS (32bits) to flash the windows 10 into the SD card. After the process completed I restarted the raspberry with the new card and after some time, it starts installing the windows OS.  I tried this process several times with different options but the result is always the same, it fails when 'Applying the image to B:'
[image]

Botspot, 2024-01-08:
The core error message here is "the semaphore timeout period has expired."
This has happened before to other users, and it seems that using better storage hardware will fix it.

Please read through this similar issue on github for more info: https://github.com/Botspot/wor-flasher/issues/40
I am not able to help any further than this, so closing the issue. However, feel free to reply back here if you have any questions or updates to share.

closed 2024-01-08 by Botspot

sayuso99, 2024-01-08:
I tried doing the same with a better SD card, and it's working properly, thank u very much ;)