Botspot/wor-flasher #142: PE-based installer integrity check failed

wuzzlefump, 2024-01-15:
## Setup:

- Which version of Windows you chose
Windows 11
- Raspberry Pi model
Raspberry Pi 400
- Name of SD card you tried to flash (some are too slow or too small)
32 Gb Sandisk Flash Drive
- What day you first downloaded wor-flasher
January 15, 2024
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
Raspberry Pi Os 64 bit

## Procedure:
1.) I installed the WoR-flasher using Pi Apps
2.) I ran the Wor Flasher gui
3.) I chose to use my flashfdrive
4.) I chose english
5) I chose to flash windows 11
6) I kept all of the default flash settings and chose to flash

I received this error:

Downloading WoR PE-based installer from Google Drive
Done ———————————————————————————————————————————————————————————————————————————
PE-based installer integrity check failed

baconstaken, 2024-01-15:
Also having this issue. The file it downloads is incomlete, only a few KBs.
I managed to find the link it was trying to download from google drive, opening it in the web browser it says google cant scan it for virus' and to click a button to continue downloading. This may be the cause for not completing the download?

wuzzlefump, 2024-01-15:
if [ "$PE_INSTALLER_SHA256" != "$(sha256sum "$PWD/WoR-PE_Package.zip" | awk '{print $1}' | tr '[a-z]' '[A-Z]')" ];then
    error "PE-based installer integrity check failed"
  fi
  
  looks like this is the line of code in the installer where things are failing

baconstaken, 2024-01-16:
I hacked my way around this and got it to start flashing, but ran into other errors possiblly unrelated. Ended up flashing wor from window OS via https://worproject.com/downloads

To hack around it: 
If you find the file 'WoR-PE_Package.zip' you see its corrupt and cant be opened. change its permissions so it cant be deleted or overwitten by anyone.
Run the wor script again and when it gets to the download it will fail and tell you the link it was trying to download.
go to that link and download the zip. 
rename it WoR-PE_Package.zip and replace the corrupt one.

Then remove lines 800-826 from the install-wor.sh script. (this will skip downloading the zip and use the one already there)

Run the wor script again and hopefully it it starts flashing.

note; i dont actually know what im doing, just fumbling my way through :)

baconstaken, 2024-01-16:
hold up, only remove 801-814
after that it extracts the zip to /peinstaller

Which was probably why it stopped working for me. Again, no idea what im doing.

DylanFlame54, 2024-01-16:
having the same problem with PE-based installer integrity check failed

jbouse, 2024-01-19:
I worked around this by downloading the `WoR-PE_Package.zip` file and placing it in the `wor-flasher-files` directory, then removing/commenting out line 810 that calls the `download_from_gdrive` function. This left the sha256sum check in place while removing the actual download step.

I found the copy of the `WoR-PE_Package.zip` that was downloaded by the script was incomplete for some unknown reason, which explains why it failed the checksum. Another fix might be to check if the file exists and, if so, skip the download attempt without hacking the script to work around it. There's already a test to see if the directory exists, so logic would then say to check whether the file exists.

I'll work on a PR with this fix in mind if I find time. I'm a licensed amateur radio operator. I use this to deploy Windows 11 onto an RPI4 8GB that our club uses to run our central network-based logging software for field days, so I re-flash the system before every field day (twice a year) to prepare. I did not encounter this issue last year before the summer field day in June, but I did while preparing for the winter field day this month (January).

hudsonaikin123, 2024-01-26:
Could anyone upload the new working file app for the Raspberry pi 400?

mrusli70, 2024-01-26:
Same here i too have the same issue and does not seems to work at all now. Any way uupdump.net server always up and down.

mrusli70, 2024-02-02:
Does they fix the problem yet? I still having the same thing. I wonder how the file get corrupted.

hudsonaikin123, 2024-02-02:
No they have not fixed it yet. Ended up sending Raspberry Pi 400 back because I could not get the WoR Flasher to work.

jbouse, 2024-02-02:
I'm not sure if the issue is the file on Gdrive that it is downloading or related to the download function itself, but when I manually downloaded the file from upstream and placed it in the download directory and commented out line 810, that calls `download_from_gdrive` and left it able to confirm the checksum.

lolimjustherefornoreason, 2024-02-06:
Is it fixed yet I’m waiting for like 1 month now

Botspot, 2024-02-10:
Fixed now

Botspot, 2024-02-10:
Google Drive changed how their "would you like to download" webpage was structured.

commit 98fe612b (2024-02-10): Fix google drive downloader, hide 0B devs, fix pi3 uefi link

Closes #143 
Closes #142 
Cloises #145
closed 2024-02-10 by Botspot