Botspot/wor-flasher #143: Unable to put windows on sd card. Getting an error message

hudsonaikin123, 2024-01-26:
## Setup:

- Which version of Windows you chose | Windows 11
- Raspberry Pi model | Raspberry Pi 400
- Name of SD card you tried to flash (some are too slow or too small) | SanDisk Ultra 256GB Micro-SD
- What day you first downloaded wor-flasher | 1/25/24
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) | Raspberry Pi OS 32-bit

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

(I was using an ethernet cable if this changes anything)

1. Put the command to download "Pi Apps" in the terminal
2.  Opened Pi apps and clicked on tools and the "wor-flasher" and then clicked install
3.  Clicked on the raspberry pi icon in the launcher and then went to accessories and then I clicked on wor-flasher
4. Launched wor-flasher and then from the startup screen I clicked windows 11 and raspberry pi4/400 then I hit next
5. After it found the best windows version I chose my language (en-us) then I hit next
6. Then I hit next after the micro-sd card popped up as a storage option. I hit nest
7. Then I hit flash
8. after a few seconds I get this error message "An error has occurred:
PE-based installer integrity check failed
Exiting now." and I hit ok. Then the popup disappears and It shows the terminal with the same message.

What should I do?

mrusli70, 2024-01-26:
I too had the same problem. I cannot seems to install anything at all.

hudsonaikin123, 2024-01-26:
ok. does anyone know how to fix this?

Botspot, 2024-02-10:
Fixed now

commit 98fe612b (2024-02-10): Fix google drive downloader, hide 0B devs, fix pi3 uefi link

Closes #143 
Closes #142 
Cloises #145
closed 2024-02-10 by Botspot