Botspot/wor-flasher #146: Raspi 5 not working with W11

dmurh, 2024-02-12:
DIRECTORY: /home/dmuth/wor-flasher
DL_DIR: /home/dmuth/wor-flasher-files
fatal: detected dubious ownership in repository at '/home/dmuth/wor-flasher'
To add an exception for this directory, call:

	git config --global --add safe.directory /home/dmuth/wor-flasher
mkdir: cannot create directory ‘/home/dmuth/wor-flasher/cache’: Permission denied
Checking for internet connection... Done

(yad:16702): Gdk-CRITICAL **: 07:08:38.872: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed

(yad:16733): Gdk-CRITICAL **: 07:08:41.036: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
/home/dmuth/wor-flasher/install-wor.sh: line 223: /home/dmuth/wor-flasher/cache/getversions.php: No such file or directory
BID: 22631.2861
RPI_MODEL: 5

(yad:16792): Gdk-CRITICAL **: 07:08:41.414: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
WIN_LANG: en-us

(yad:16850): Gdk-CRITICAL **: 07:08:43.636: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 1

(yad:16874): Gdk-CRITICAL **: 07:08:49.861: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed

User chose to download everything to RAM.
For best results, please close all other programs. (especially web browsers and games)

(yad:16888): Gdk-CRITICAL **: 07:08:51.554: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
/home/dmuth/wor-flasher/install-wor-gui.sh: line 650: syntax error: unexpected end of file

Botspot, 2024-02-12:
This is a file permission issue.
Could you run this and copy the output here?
```
ls -l wor-flasher
```

dmurh, 2024-02-12:
after 👍 
fatal: detected dubious ownership in repository at '/home/dmuth/wor-flasher'
To add an exception for this directory, call:

	git config --global --add safe.directory /home/dmuth/wor-flasher
mkdir: cannot create directory ‘/home/dmuth/wor-flasher/cache’: Permission denied
Checking for internet connection... Done

(yad:2708): Gdk-CRITICAL **: 07:49:41.905: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed

(yad:2739): Gdk-CRITICAL **: 07:49:46.179: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
/home/dmuth/wor-flasher/install-wor.sh: line 223: /home/dmuth/wor-flasher/cache/getversions.php: No such file or directory
BID: 22631.2861
RPI_MODEL: 5

(yad:2798): Gdk-CRITICAL **: 07:49:46.474: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
WIN_LANG: en-us

(yad:2841): Gdk-CRITICAL **: 07:49:48.846: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 1

(yad:2864): Gdk-CRITICAL **: 07:49:51.983: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed

dmurh, 2024-02-12:
also i did   git config --global --add safe.directory /home/dmuth/wor-flasher     same result

Botspot, 2024-02-12:
No, @dmurh, you did not understand. I need you to show me the **output** from when you run `ls -l wor-flasher`.
This command does not fix anything, but it might help me understand what sort of problem you are encountering.

dmurh, 2024-02-12:
sorry ...

yad:3223): Gdk-CRITICAL **: 15:05:44.613: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed

(yad:3254): Gdk-CRITICAL **: 15:05:49.071: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
BID: 22631.2861
RPI_MODEL: 5

(yad:3313): Gdk-CRITICAL **: 15:05:49.359: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
WIN_LANG: en-us

(yad:3355): Gdk-CRITICAL **: 15:05:52.653: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 1

(yad:3378): Gdk-CRITICAL **: 15:05:56.361: gdk_x11_window_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed

/home/dmuth/wor-flasher/install-wor-gui.sh: line 650: syntax error: unexpected end of file
dmuth@Raspi4:~ $ ls -l wor-flasher
total 1072
drwxr-xr-x 2 dmuth dmuth   4096 Feb 12 15:05 cache
-rw-r--r-- 1 dmuth dmuth    992 Feb 12 15:04 config_txt_tips
-rwxr-xr-x 1 dmuth dmuth  28760 Feb 12 15:04 install-wor-gui.sh
-rwxr-xr-x 1 dmuth dmuth  49658 Feb 12 15:04 install-wor.sh
-rw-r--r-- 1 dmuth dmuth   8052 Feb 12 15:04 logo-full.png
-rw-r--r-- 1 dmuth dmuth   3117 Feb 12 15:04 logo.png
-rw-r--r-- 1 dmuth dmuth  66469 Feb 12 15:04 next-steps.png
-rw-r--r-- 1 dmuth dmuth 749931 Feb 12 15:04 next-steps.svg
-rw-r--r-- 1 dmuth dmuth  11895 Feb 12 15:04 overview.png
-rw-r--r-- 1 dmuth dmuth 128113 Feb 12 15:04 overview.svg
-rw-r--r-- 1 dmuth dmuth   1692 Feb 12 15:04 ram.png
-rw-r--r-- 1 dmuth dmuth  12762 Feb 12 15:04 README.md
-rwxr-xr-x 1 dmuth dmuth   2378 Feb 12 15:04 terminal-run

Botspot, 2024-02-12:
OK try running this command to fix the permissions:
```
chown -R dmuth:dmuth ~/wor-flasher-files
chown -R dmuth:dmuth ~/wor-flasher
```
If either of those give you any errors, run them again with sudo.

dmurh, 2024-02-12:
thanks

i see  no folder  /home/dmuth/wor-flasher-files
 --   create it and  - now it starts to run

will come back tomorrow with result

Botspot, 2024-02-12:
OK I think I know what you did. When downloading wor-flasher, did you happen to use sudo?
```
sudo git clone https://github.com/Botspot/wor-flasher
```
If you used sudo, then that is your problem. You should not use sudo for this.

Botspot, 2024-02-12:
Anyway, the chown commands you ran earlier fixed this so you should be good to go.

closed 2024-02-12 by Botspot

dmurh, 2024-02-12:
i think it was the missing folder - nevertheless  - i will test within the week again and come back 
- and i am not sure , but i don`t use sudo