Botspot/wor-flasher #147: PE-based installer integrity check failed

dbtronics, 2024-02-13:
## Setup:

- Which version of Windows you chose: 11
- Raspberry Pi model: 4 Model B
- Name of SD card you tried to flash (some are too slow or too small): Lexar USB Flash Drive 512GB
- What day you first downloaded wor-flasher: February 12, 2024
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc): Raspberry Pi OS Bullseye (But also tried Bookworm as well)

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

This is my terminal output:

Choose Windows version:
1) Windows 11
2) Windows 10
3) More options...
Enter 1, 2 or 3: 1

Finding newest build...
Selected version: Windows 11 build 22631.2861 

ar-sa - Arabic (Saudi Arabia)
bg-bg - Bulgarian (Bulgaria)
cs-cz - Czech (Czechia)
da-dk - Danish (Denmark)
de-de - German (Germany)
el-gr - Greek (Greece)
en-gb - English (United Kingdom)
en-us - English (United States)
es-es - Spanish (Spain, International Sort)
es-mx - Spanish (Mexico)
et-ee - Estonian (Estonia)
fi-fi - Finnish (Finland)
fr-ca - French (Canada)
fr-fr - French (France)
he-il - Hebrew (Israel)
hr-hr - Croatian (Croatia)
hu-hu - Hungarian (Hungary)
it-it - Italian (Italy)
ja-jp - Japanese (Japan)
ko-kr - Korean (Korea)
lt-lt - Lithuanian (Lithuania)
lv-lv - Latvian (Latvia)
nb-no - Norwegian Bokmål (Norway)
nl-nl - Dutch (Netherlands)
pl-pl - Polish (Poland)
pt-br - Portuguese (Brazil)
pt-pt - Portuguese (Portugal)
ro-ro - Romanian (Romania)
ru-ru - Russian (Russia)
sk-sk - Slovak (Slovakia)
sl-si - Slovenian (Slovenia)
sr-latn-rs - Serbian (Latin, Serbia)
sv-se - Swedish (Sweden)
th-th - Thai (Thailand)
tr-tr - Turkish (Turkey)
uk-ua - Ukrainian (Ukraine)
zh-cn - Chinese (Simplified, China)
zh-tw - Chinese (Traditional, Taiwan)

From the list above, enter a language: en-us

Choose Raspberry Pi model to deploy Windows on:
1) Raspberry Pi 5
2) Raspberry Pi 4 / 400
3) Raspberry Pi 3 or Pi2 v1.2
Enter 1, 2, or 3: 2

Available devices:
/dev/sda - 461.3GB - Lexar USB Flash Drive
Choose a device to flash the Windows setup files to: /dev/sda

Would you like to:
1) Create an installation drive capable of installing Windows to itself
2) Create a recovery drive to install Windows on other >16 GB drives
Choose the installation mode (1 or 2): 1

Input configuration:
DL_DIR: /home/dianbasit/wor-flasher-files
RUN_MODE: cli
RPI_MODEL: 4
DEVICE: /dev/sda
CAN_INSTALL_ON_SAME_DRIVE: 1
BID: 22631.2861
WIN_LANG: en-us

Downloading WoR PE-based installer from Google Drive
Done ————————————————————————————————————————————————————————————————————————————————————————————————————————————————
PE-based installer integrity check failed

I tried flashing this yesterday on my Pi 4, and it flashed fine. Then today, I tried flashing and it was continuously giving me integrity check error. I tried this on bookworm and bullseye version of RPi OS and still couldn't get through this error. I even reformatted my usb many times, but couldn't resolve this issue at all.

Please let me know what I should do to resolve this issue.

Thanks.
[image]

Botspot, 2024-02-13:
Looks like the google drive downloader broke again.

Botspot, 2024-02-13:
Oh Google actually blocked the file because so many users were downloading it! :)
[image]

Leaf239, 2024-02-13:
me toooo!! @Botspot , please fix for me!! (windows 10 on raspi.5)

Leaf239, 2024-02-13:
@Botspot .. please 😭

Leaf239, 2024-02-13:
nobody fix it.... lol

Botspot, 2024-02-13:
I'm trying to get around google's restrictions as we speak...

Botspot, 2024-02-13:
If that doesn't work then I will have everybody download the PE package from github instead.

dbtronics, 2024-02-13:
So @Botspot after 24 hrs, the link should be operational then?

Leaf239, 2024-02-13:
@Botspot , thank you so much!!

Leaf239, 2024-02-13:
Could you please upload the Windows 10 arm64 file(iso) to GitHub now? I want to download it right away.

Leaf239, 2024-02-13:
so, Do I have to wait 24 hours to download it?

Botspot, 2024-02-13:
OK this should be fixed now. Google Drive is no longer used. See what I did in this code change: https://github.com/Botspot/wor-flasher/commit/abcc5a3248892061517d72a0ec4befa940cd1ab7

Just run wor-flasher again and everything should work now.

closed 2024-02-13 by Botspot

dbtronics, 2024-02-13:
Yes issue resolved! Thanks