Botspot/wor-flasher #148: windows 11 on raspberry pi 5: NO wifi or wiried connection

Leaf239, 2024-02-13:
i just use wor installer (https://github.com/Botspot/wor-flasher), but i cannot connect internet. as Diagnostic results, somthing wrong with driver.

Botspot, 2024-02-13:
This is a known issue and I have written about it on the "Useful information" section.

Did you read that?
[image]
If you did not see that earlier, can you think of a better place for me to put the information so that people like you will actually read the important information?

Leaf239, 2024-02-14:
oh, I understand. there has no driver for raspi 5. Thank You!

Leaf239, 2024-02-14:
When do you plan to update the driver for the Pi 5 so that it can be installed, @Botspot ?

Botspot, 2024-02-14:
Only the wifi module manufacturer could, and they would need help from microsoft. That will never happen.

closed 2024-02-14 by Botspot

Leaf239, 2024-02-14:
Never able to use the internet?

Botspot, 2024-02-14:
Use a USB to Ethernet adapter.

Leaf239, 2024-02-14:
Ok. I'll buy USB Adapter on Amazon.

Botspot, 2024-02-14:
Or pair your phone to it the way leepspvideo did here: https://www.youtube.com/watch?v=haeYUq4RHbk

Leaf239, 2024-02-23:
Thanks! But I've already bought an adapter and I'm using it. LOL

Leaf239, 2024-02-23:
What a nice way!

Goldfelty, 2025-06-12:
I imagine there could be a way to take a driver form the galaxy S2 1st ultra which uses the same chip and format this into something windows can understand