Botspot/wor-flasher #150: dgk_x11_window_get_xid

vcesinger, 2024-04-03:
## Setup:

- Windows10
- Raspberry Pi 5
- NVME Device 512GB
- 04/03/24
- Raspberry Pi OS

The installscript ends with :
yad:26876: Gdk-CRITICAL ** gdk_x11_windows_get_xid: assertion 'GDK_IS_X11_WINDOW (window)' failed

The installation does not work.

Botspot, 2024-04-03:
That warning is normal on PiOS Bookworm and any Wayland system, is non-fatal, and should not affect functionality.

> The installation does not work.

If you want help with that, then you will need to be specific on what actually happened.

vcesinger, 2024-04-05:
Hi, 
thanks for getting back to me. I thought this error was the problem, that i can not install it on a SSD. I am using a GeeekPi N05 M.2 M-Key NVMe HAT für Raspberry Pi 5 with a 512 GB SSD.
Installation of PI OS works with no errors, the WOR-Flasher results in a missing boot device.
Any idea?

Botspot, 2024-04-05:
> Hi, thanks for getting back to me. I thought this error was the problem, that i can not install it on a SSD. I am using a GeeekPi N05 M.2 M-Key NVMe HAT für Raspberry Pi 5 with a 512 GB SSD. Installation of PI OS works with no errors, the WOR-Flasher results in a missing boot device. Any idea?

Did wor-flasher actually **give you an error?**
Or did you encounter an error after booting the drive flashed with wor?

vcesinger, 2024-04-06:
Hi, i made a step forward. I discovered that the WOR-Flasher uses the 0.2 UEFI version which does not support NVME, now i managed to get it to 0.3 but after starting the windows installer is returning an error that it misses a drive to install to.

Botspot, 2024-04-16:
I've updated wor-flasher to use the 0.3 version of Pi5 UEFI. Any updates on this?

Duuckjing, 2024-04-18:
I tried this and the usb got three partitions,one is the uefi, the second is 18GB partition with a file "install.wim", but the other one can't be seen even in diskgenius.

vcesinger, 2024-05-08:
I am facing the same issue...

SeaDog58, 2024-05-25:
BotSpot,
Some weeks ago I used WOR Flasher to create a Windows Pro 11 installation on a Raspberry Pi 5 on a 500 gb USB drive successfully.
Now I have been trying the same installation process to put Windows Pro 11 on an NVME drive.  I have been unsuccessful.  Here are the hardware details.
•	Raspberry Pi 5 running Raspberry OS on a 32gb SD
•	Hat: GeeekPi N04 M.2 NVMe to PCIe Adapter for Raspberry Pi 5, Support M.2 NVMe SSD
•	SSD: Western Digital 1TB SN750 NVMe Internal Gaming SSD Solid State Drive - Gen3 PCIe, M.2 
I had been getting the critical error on GDK as reported above but now understand that is not an issue.
I was having problems with not being able to find a boot device.  Since you reported you have updated wor-flasher to use the 0.3 version of Pi5 UEFI, I used “git clone https://github.com/Botspot/wor-flasher” today to update my files.
I ran the gui version of Flasher.  I selected the NVME drive for flashing Windows Pro 11 and it completed without obvious errors.  After flashing and rebooting, I noted in the Raspberry OS desktop, it showed an 18gb drive with a file install.wim.
After that, I waited a long time, and nothing happened. So I did sudo raspi-config and selected option 2 for booting from NVME, USB, and SD in that order.  Rebooted.
After some time, WOR-PE reported it couldn’t find the installation disk.  The log showed: 
•	Disk 0 now the selected disk.
•	Virtual disk service error.
•	The disk not initialized. Exit code 2147212265
•	Couldn’t mount the partition.
•	Couldn’t fine the PE disk
After flashing should I just let it sit there without rebooting?  Could that be the problem?
John

chrisgch, 2025-01-09:
I have the same problem with WoR from the pi-apps store. I have installed Windows 11 from Raspbian to the official 256GB Raspberry NVMe SSD, which ended with the same 'GDK_IS_X11_WINDOW (window)' failed error. I then tried booting from the SSD, which resulted in error "Could not find the installation disk!". The log is the same as reported by SeaDog58 in May. Any ideas?

chrisgch, 2025-01-10:
Another test with the PI5: I installed WoR to a USB flash drive and booted from that. This let me choose the NVMe drive as installation target, but unfortunately dispart then failed with an error. The log file is attached. I also tried replacing storahci.sys and stornvme.sys in boot.wim from the download in [this post](https://github.com/worproject/rpi5-uefi/issues/2#issuecomment-2016505218)  but got the same results.
[WoR-PE.log](https://github.com/user-attachments/files/18380253/WoR-PE.log)

SeaDog58, 2025-01-22:
> I have made a little progress: I installed WoR to a USB flash drive and booted from that. Then I connected the NVMe drive in an USB enclosure and installed to that. This worked fine, Windows 11 24H2 is running very fast from the USB SSD. I then tried to connect the NVMe drive via adapter to the Pi5, but this did not boot (INACCESSIBLE_BOOT_DEVICE). Again I tried to use the drivers from the previous post, but that didn't work either. So for now I will continue to use the USB SSD enclosure.

Chris,
I have also have trouble with NVME drives.  When you say adapter, I assume you are talking about a HAT for the RPi.  Right now, I am also getting INACCESSIBLE_BOOT_DEVICE on the RPi5 that was working.  It is strange that it happens periodically.  I think there is a driver issue of some sort.  Window 11 on RPi is pretty clearly not ready for prime time.
John