Botspot/wor-flasher #151: No boot on rp5

Duuckjing, 2024-04-15:
## Setup:

- Which version of Windows you chose?  Win10
- Raspberry Pi model. rpi5
- Name of SD card you tried to flash (some are too slow or too small). SanDisk ultra 128gb
- What day you first downloaded wor-flasher. Last day
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc).  Raspi os

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
  
I followed the steps on the readme file: plugging in the SanDisk card via a tf card reader through USB, I first used git to clone the wor-flasher, and then launched the GUI, after waiting for it to be flashed, I closed the window and pull out the kioxia tf card on which installed my raspberry pi os. Finally I plugged in the SanDisk and wait it to boot, but nothing happened. I have checked it on my phone, the card's status seems to like this:[image]

Botspot, 2024-04-15:
Try Windows 11.

Duuckjing, 2024-04-18:
Currently I have tried various methods, but windows couldn't be seen installed on raspberry pi through tf card. Now the only way seems to be booting through usb2.0 https://www.bilibili.com/read/mobile?id=33618795

Botspot, 2024-04-18:
TF card does work. If the TF card is not working for you, try a different one.

Duuckjing, 2024-04-18:
yep， I believe in that, and I'm trying to figure it out