Botspot/wor-flasher #152: Installing from USB to NVMe

Niehztog, 2024-04-16:
Hey, I have a Raspberry Pi 5 with attached NVMe. I would like to install Windows 10 or 11 on my NVMe (dualboot with Raspberry Pi OS would be ideal). I already have Raspberry Pi OS installed on my NVMe with plenty of unallocated disk space left.

I downloaded wor flasher, and flashed the windows 10 installation files on to my 16 GB USB stick. When booting from that stick I get into the Windows PE Installer where I can select the destination drive. But mv NVMe is not appearing, I only see the USB stick. I added `dtparam=pciex1_gen=3` with the same result.

Should the PE installer support NVMe drives yet? Does it support installing Windows onto a drive which already has another OS installed?

Botspot, 2024-04-16:
My first suggestion is to try Windows 11 with the Pi5.
Second suggestion: try again.
Third suggestion: don't attempt to do dual boot. Pretty sure that is not really doable unless you can write code.

I just checked the Pi5 UEFI github releases page and version 0.3 says it added support for NVME and SATA support. See the release notes for yourself: https://github.com/worproject/rpi5-uefi/releases/tag/v0.3

I was unaware of the release of 0.3 until now, so until a few minutes ago WoR-Flasher was still using v0.2.

To get this fix, simply run wor-flasher again which should update its script. But the uefi folder containing 0.2 files will still be there, preventing the download of uefi version 0.3. So delete `~/wor-flasher-files/pi5-uefipackage`, then follow the flashing process again.

Closing this issue as it ought to work now. Even though it is closed, please comment here to let me know how it goes.

Niehztog, 2024-04-17:
Hi @Botspot , thank you for the fast response. Your fix seems to work. I flashed another USB stick with windows 11 and then I saw the NVMe in the PE installer. Thank you.

Any ideas on how to accomplish dual boot with another Raspberry Pi OS partition?