Botspot/wor-flasher #155: Looking for Disk to Install Error

Potatz, 2024-05-14:
## Setup:

- Which version of Windows you chose - Windows 10
- Raspberry Pi model - Raspberry Pi 4B
- Name of SD card you tried to flash (some are too slow or too small) - SanDisk 256gb
- What day you first downloaded wor-flasher - 5/13/24
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) - Raspberry Pi OS

The Raspberry Pi is connected to an Ethernet cable... Don't know if that's needed or is interfering but thought it could be important.

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
(I used this guide for installing windows: https://www.tomshardware.com/how-to/install-windows-11-raspberry-pi)
 Ran: git clone https://github.com/Botspot/wor-flasher
 Then ran: ~/wor-flasher/install-wor-gui.sh
 Installed on a 7.5gb thumb drive
 Shutdown Pi
 Removed SD card w Raspberry Pi OS
 Started up again with thumb drive
 Everything started normally, during the loading part (and it went pretty fast).
 Then it changed to a fully blue screen (not blue screen of death) and it looked for a disk to install on. Dunno what it meant by that but assumed that it would use the thumb drive (it didn't) and ended up giving me an error. 
Error said: _Couldn't find the installation disk! Make sure it is connecting to your device and has all the files necessary, then click on the reload button. If the error persists please check the log for more details._
My mouse and keyboard didn't work so i just unplugged it.

 After that i tried removed Raspberry Pi OS from the SD card and put it in, I assumed maybe it wanted an SD card to install on?, did the same thing and gave me an error .-.

During startup it will also try and start using IPv4, IPv6, HTTP boot IPv4, and HTTP boot IPv6 which all fail before it goes to the blue screen and the looking for drive...

I'm pretty new to this, so I could be doing something horribly wrong .-.

Potatz, 2024-05-14:
I am currently trying to install windows 11 onto the thumb drive instead, to see if that works any better. I'm using the exact same steps, except I'm using windows 11

Botspot, 2024-05-14:
Other people have seen this error, though I never have.
The main 2 solutions are:

1. Get a faster storage device. Windows is known to not like booting from any old thumb drive. This is not a problem specific to Raspberry Pi.
2. Update your Pi's bootloader. Really old bootloaders seemed to have this problem. You can do this from within Pi OS or flash a bootloader recovery image to the sd card using Pi imager.

Potatz, 2024-05-14:
Kay. Two things:
A: Tried installing Windows 11 and it broke for some reason? Also now the thumb drive is write protected and I can't fix that .-.
More importantly, B: How do I update the bootloader? I trying to re-install Pi OS right now, dunno if that'll do the trick.

Botspot, 2024-05-14:
> Kay. Two things: A: Tried installing Windows 11 and it broke for some reason? Also now the thumb drive is write protected and I can't fix that .-.

Some drives will do that after long enough time or if they lose power during a bunch of writes. There are tricks shared online to try to un-lock them but results can be hit or miss.

> More importantly, B: How do I update the bootloader? I trying to re-install Pi OS right now, dunno if that'll do the trick.

https://pimylifeup.com/raspberry-pi-bootloader/

Potatz, 2024-05-16:
Scrolled down to another way of installing windows (on the same tutorial), and am trying that right now (USB didn't want to un-write protect). So far everything is going well, but I cant access www.worproject.ml? I dont know why... But it just keeps saying timed out. Did you guys discontinue the site or something?

Botspot, 2024-05-16:
Use www.worproject.com

Tigerman2309, 2024-10-22:
I have a pi400, and was interested in trying Windows On Raspberry

I tried a USB dock for a M2 drive, it worked with the Windows Version, but fails with the Raspberry PI Version of the software, with "WoR-PE - Couldn't find the installation disk!"

doughnet, 2024-11-19:
Same here with Pi4.  Windows 11 won't even boot.  Windows PE loads but says failed to install.  Tried on USB NVME and USB SSD.

Same Pi4 was able to install using worproject.com but it fails using this wor-flasher.