Botspot/wor-flasher #159: [Question] What versions of Windows can you install?

Panda2340, 2024-06-19:
Hello,
I was just wondering, what versions of Windows can you use the tool to install? I was looking to install Windows XP or Vista on a machine and wanted to see if I could use this tool to get it on my Raspberry Pi 400.

Botspot, 2024-06-19:
Just windows 10 or 11. Older versions of windows never had an ARM release, so they cannot directly run on Raspberry Pi, which has an ARM CPU instead of an x86 CPU architecture.
However, you can run XP or Windows 95 using the QEMU emulator. Look on YouTube for how to do this.

closed 2024-06-19 by Botspot

Panda2340, 2024-06-19:
> Just windows 10 or 11. Older versions of windows never had an ARM release, so they cannot directly run on Raspberry Pi, which has an ARM CPU instead of an x86 CPU architecture. However, you can run XP or Windows 95 using the QEMU emulator. Look on YouTube for how to do this.

Oh, makes sense. I will look in to QEMU, and I might get Windows 10 or 11 on my Raspberry Pi 400. Thank you.

Panda2340, 2024-06-21:
> Just windows 10 or 11. Older versions of windows never had an ARM release, so they cannot directly run on Raspberry Pi, which has an ARM CPU instead of an x86 CPU architecture.
> 
> However, you can run XP or Windows 95 using the QEMU emulator. Look on YouTube for how to do this.

Oh, that makes sense. I will look into QEMU, and I might get Windows 10 or 11 on my Raspberry Pi 400.