Botspot/wor-flasher #160: Pi hat fan on raspberry pi 4b not working

Xavierzon, 2024-07-13:
## Setup:

- Which version of Windows you chose: latest version
- Raspberry Pi model: 4B 4gb
- Name of SD card you tried to flash (some are too slow or too small): 64 gb sandisk
- What day you first downloaded wor-flasher: 3 weeks ago
- 
## Procedure:
I want to make the fan on the **Raspberry Pi PoE+ HAT** work. only there is (in my eyes) no way to configure the fan speeds ect.. the fan wont run in any other way. even though i used fan control. There was no way to detect the fan or to configure the fan speeds.

Botspot, 2024-07-13:
If the POE hat can be configured with config.txt then this is easy. Otherwise you're out of luck unless you're ready to learn coding at a level that even I don't understand.

Xavierzon, 2024-07-13:
I do know how to configure this in a debian os with config.txt but i do not know (and i dont think so) windows 11 iot has a config.txt for that... the same issue for me here that i have no clue how to configure that in windows...

Botspot, 2024-07-13:
Windows 11 has a config.txt so just copy the lines over from your working system.

Xavierzon, 2024-07-13:
Do you have the location to the file? i cannot find it in the iot version..

Botspot, 2024-07-13:
Are you using wor-flasher? It has a section for adding lines to config.txt.

Xavierzon, 2024-07-13:
i have used it... is there a way to do it afterwards? otherwise i have to reinstall and try with the default lines from a debian pc.

Botspot, 2024-07-13:
> i have used it... is there a way to do it afterwards? otherwise i have to reinstall and try with the default lines from a debian pc.

You should be able to shutdown windows, remove the sd card, and use debian to read it. The config.txt should just be there. You cannot edit config.txt from within windows.

Xavierzon, 2024-07-13:
I do not have a dual boot.. long story short i can not find a way to make te pi hat fan work with windows 11 iot...

closed 2024-07-13 by Xavierzon

Botspot, 2024-07-13:
Do you only have 1 sd card?

It should be easy to boot a debian sd card in order to read the windows sd card.
This takes 2 sd cards, 1 raspberry pi, and no dual boot.

Xavierzon, 2024-07-13:
I understand i made boot from a thumbdrive... but when i do that i get the usual C drive without config.txt. i have no clue where that config.txt should be on the windows sd.

Botspot, 2024-07-13:
> I understand i made boot from a thumbdrive... but when i do that i get the usual C drive without config.txt. i have no clue where that config.txt should be on the windows sd.

There should be 2 partitions and the config.txt is on the boot partition.