Botspot/wor-flasher #163: Constantly get: The wimupdate command failed to add peinstaller to boot.wim

phuongnd08, 2024-09-12:
## Setup:

- Which version of Windows you chose: Windows 11
- Raspberry Pi model: 5
- Name of SD card you tried to flash (some are too slow or too small): SamSung Evo Plus (130M/s)
- What day you first downloaded wor-flasher: yesterday
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc): Linux ubuntu2204 5.15.0-119-generic #129-Ubuntu SMP Fri Aug 2 20:37:01 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux

## Procedure:
The linux system is a virtual one running on UTM. Attach USB SDcard Reader to Machine.
Step 1: Clone https://github.com/Botspot/wor-flasher
Step 2: Run ./install-wor.sh
Step 3: Choose Windows 11, en-us, pre-installer
Step 4: Getting error

phuongnd08, 2024-09-12:
Error log:

```
Copying files to /dev/sda:
  - Startup environment
  - Installation files
`  - EFI files
  - PE installer
The wimupdate command failed to add /home/phuongnd08/wor-flasher-files/peinstaller to boot.wim
Errors:
[ERROR] "/media/phuongnd08/wor-flasher/bootpart/sources/boot.wim": Error reading header: Input/output error
ERROR: Exiting with error code 50:
       Could not read data from a file.
```

WombleWoo7547, 2025-01-12:
From my understanding, this appears to be an issue where the Linux host cannot read the download boot.wim file. Try redownloading Windows 11 through wor-flasher, re-installing the UTM linux host, and if all else fails, use the Pi to image the Pi.

Most Pi 4s and all 5s have RPi-imager built in to the bootloader. Flash RPiOS to that and then use WoR flasher to image a USB stick with the PE installer, then use that to install onto the MicroSD. That would be the ideal solution to the problem, that's **nearly** always going to work (always nearly with computers :D).

As Linus Sebastian from the Tech Tips himself said: "Windows is a piece of s**t. Linux is a piece of s**t ... MacOS is also a piece of s**t"