Botspot/wor-flasher #169: Installing issue

PIETRO69d, 2024-11-08:
## Setup:

- Which version of Windows you chose
- Windows 11 pro
- Raspberry Pi modepi5
- Name of SD card you tried to flash (some are too slow or too small)
- fanxiang s500 pro l (nvme)
- What day you first downloaded wor-flasher
- 3 November 2024
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc)
- Pi os 64bit

## Procedure:
I downloaded wor flasher via pi apps
I flashed it on fanxiang s500 pro
Turned it off and booted in the preinstallation environment
I tried to install win11, but it gives me this errors
[image]
[image]

Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

Marcinoo97, 2024-11-17:
NVMe is not actually supported be default.    requires few files to be replaced to be fully supported.

vaccinemedia, 2024-12-06:
I have the same issue here and was installing on to a USB drive connected to the pi:
[image]
[image]

ajmeese7, 2024-12-07:
Same issue, does not work at all on a Pi 5 with Windows 11:

[image]

[image]

vaccinemedia, 2024-12-07:
I managed to get through the installation process by creating an iso image which does not include the updates. The iso was smaller as a consequence and installed a bit faster too. On the Windows installation wizard at one point it asks you to connect to the internet. At the time I did not have any Pi5 accessories so I hit shift+F10 to get a command prompt and then ran OOBE\BYPASSNRO - this then. restarts the Pi and gets you back on the wizard but the difference being that there's now a "I don't connect to the internet" ption which got me through the installation / account creation process and onto the Windows 11 desktop.

I ordered a TP-Link USB Ethernet adapter from Amazon [](https://www.amazon.co.uk/dp/B09MFY8799?ref=ppx_yo2ov_dt_b_fed_asin_title) and this is recognised automatically. I'm about to run Windows update and see if everything updates OK but as far as this issue is concerned I did manage to get to the desktop by using an iso without updates included.

closed 2025-01-07 by PIETRO69d