Botspot/wor-flasher #171: Issue: No boot after error free installation.

nasyx-rakeeb, 2024-12-14:
## Setup:

- Which version of Windows you chose: Latest 
- Raspberry Pi model: 4B
- Name of SD card you tried to flash (some are too slow or too small): Samsung Evo 64 GB
- What day you first downloaded wor-flasher: Today
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc): Raspberry Pi Os 64 Bit

## Procedure:
I installed the latest version available in wor flasher of windows 11 and the process went smoothly without any errors, and after i tried to boot it it's just stuck at that initial boot screen that shows up where that big raspberry pi logo is shown for reference i have attached the image below.
[image]
It's been stuck there for hours and nothing happens. I have tried both usb drive and sd card but still same issue. I have installed windows 11 via wor-flasher from Pi-Apps before on both sd card and usb Drive Maybe a year ago and there were no issues but now i have this issue and don't know whats causing it.

mikekolling1966, 2024-12-17:
mine doesnt even get to the boot screen, just blank. and no errors when creating the flash to the sd card.

Marcinoo97, 2024-12-19:
Tool selected version of windows 11 that is too new for pi 4.

Mortando, 2024-12-24:
> mine doesnt even get to the boot screen, just blank. and no errors when creating the flash to the sd card.

Try modifying the parameter framebuffer_depth=24  (by default it's 32) in the config file. (either when running the wor-flasher and you get to the custom parameters part ... or after by connecting the already installed SD into another machine or raspberry ... go to the 999 directory and modify in the config file there).

Marcinoo97, 2024-12-24:
> Try modifying the parameter framebuffer_depth=24 (by default it's 32) in the config file. (either when running the wor-flasher and you get to the custom parameters part ... or after by connecting [...]

your suggestion is applicable to pi 5 only. You don't know what pi board he has.

MariuszFerdyn, 2024-12-26:
To be checked - 22631.2861version should work for the Pi 4. Follow #166