Botspot/wor-flasher #174: Speed issue

vaisd007302, 2025-01-16:
## Setup:

- Windows 11
- RPi 40
-USB2 64GB Drive
- 8th Sep 2024
- Raspberry Pi OS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.
I selected pi 4/400, selected more options, choose a custom windows version, selected 22631.2861. I did not edit config.txt at all.

When i booted i waited 8 hours but it was displaying 'Installing WIndows 11' (12%).
I tried again but got the same result.

Marcinoo97, 2025-01-16:
I'm guessing that your storage device is either worse quality or almost dead

vaisd007302, 2025-01-17:
I tested it using dd and it has write speeds of about 27MB/s. thats megabytes not megabits.