Botspot/wor-flasher #178: Pi5 not booting from nvme: Stock on recovery screen

DipuGit-creator, 2025-03-02:
The pi boots properly when the nvme is connected through an adapter to an usb port, but when using a PCIe hat, the boot gives me a BSOD and later sends me to recovery mode with the error:

`The application or operating system couldn't be loaded because a required file is missing or contains errors.`

File is : winload.efi (The System32 one) 
Except, it IS present 

The BSOD Stop Code is:
INACCESSIBLE_BOOT_DEVICE

Marcinoo97, 2025-03-02:
expected behaviour if this project wants to remain legal.

DipuGit-creator, 2025-03-02:
> expected behaviour if this project wants to remain legal. those files from https://cdn.discordapp.com/attachments/1082848823233216532/1213621523793842257/storage_v4.zip?ex=67c58b5f&is=67c439df&hm=5e1680a7e3d113583af94546feddcd109ac969600160e28a947a2504ca7af2b4& need to be replaced in System32 with file permissions overwritten.

So I have to download this file and put it to System32?

DipuGit-creator, 2025-03-03:
Hello?

Botspot, 2025-03-03:
Maybe try BVM

Botspot, 2025-03-03:
Either that or look around on youtube for tutorials. I don't use wor or wor-flasher myself so I can't really keep it running well.