Botspot/wor-flasher #179: Windows 11 installation failing on RPi 5

anniesnn, 2025-03-14:
## Setup:

- Which version of Windows you chose 11
- Raspberry Pi model RPi5 8 GB
- Name of SD card you tried to flash (some are too slow or too small) RPi SSD 512 GB
- What day you first downloaded wor-flasher 13-Mar
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) RPi OS

## Procedure:
Tell us what you did, step by step. Be specific enough so that one of us would be able to replicate your setup.

I followed the online videos and installed the flash on raspberry pi. However, when I connect the SSD to start the windows installation, it fails with this error:
```
Couldn’t find the installation disk. 
Make sure it is connected to your device and has all the files necessary then check on the reload button
If the error persists, please check the log for more details. 
```

I can see this error in the log:
```
Disk 0 is now the selected disk.
Virtual Disk Service error.
The disk is not initialized,
2024-04-01 01:05:31.1890 [warn] Exit code: 2147212265
2024-04-01 01:05:31.2048 [warn] Couldn't mount the partitions!
2024-04-01 01:05:31.2195 [error] Couldn't find the PE disk!
```

Could you please advise what is wrong going on here and how to fix this issue? 

Thank you very much, in advance.

Marcinoo97, 2025-03-14:
NVMe will not be supported as project needs to remain legal and system files in windows have a problem with pi's broken pcie implementation. Modified files exist to fix this issue but project can't include them. you are free to use them manually tho. 

[storage_fix8.zip](https://github.com/user-attachments/files/19252604/storage_fix8.zip)

anniesnn, 2025-03-15:
Hi

anniesnn, 2025-03-15:
Apologies, I closed by mistake. 

Hi @Marcinoo97 - could you please assist how to apply these patches, even before the installation can take place?

Marcinoo97, 2025-04-03:
no