Botspot/wor-flasher #180: Windows 11 Pi 5 color looks off and stuck on bootscreen

hassanm12, 2025-03-15:
## Setup:

- Which version of Windows you chose
- Raspberry Pi model = 5
- Name of SD card you tried to flash (some are too slow or too small) = 64gb but installed Windows on external SSD
- What day you first downloaded wor-flasher - Today
- What linux operating system you are using to run wor-flasher on. (Raspberry Pi OS, Twister OS, Debian, etc) = Pi OS 64bit

## Procedure:
I installed windows 11 using the GUI. If framebuffer=32 is in my config file, it just shows a black screen. If I delete or comment out that line, the boot screen is green and purple

[image]

Anyway, now it is stuck on that screen for about an hour. 

Anyone know if I am doing something wrong?

Marcinoo97, 2025-03-15:
how much ram?

hassanm12, 2025-03-15:
8GB

hassanm12, 2025-03-15:
> how much ram?

8GB

PabloIvnGl, 2025-04-16:
same error here, rpiEFI. so its a model D problem, this model doesn't support or boot the efi. if any have a solution comment pls

NumberOneGit, 2025-05-02:
There are two problems with RPI5-UEFI on D0:

1)   GPIO was totally reworked internally for the D0, which was part of why it was created in the first place.   Pi Foundation wanted to get rid of “dark silicon” and there were tons of pins that weren’t being used for anything by the C1 model that made the chip consume more power/less efficient.   All of the external pins remain the same and I believe all of the physical memory addresses stay the same, but to bridge the differences between the C1 and D0, there needs to be a Pinctrl file.   If you look in the Linux source files, you will see that a file exists that references both chips and fixes all routing issues.  In RPI5-UEFI, all Pinctrl is hard-coded to C1 values.   The ideal solution is to use the code that already exists that checks board revision/model for other functions and include that in the Pinctrl files, with both datasets in the file and the board revision deciding which one to use based on the board attached.  The other option, which is the one I’m using now because I don’t know edk2 codebase very well, is to just replace all the Pinctrl values with the D0 ones and re-compile.

2)  Framebuffer does seem to work “overall”.  If framebuffer_depth is set to 16 or 24 bits, the image looks distorted in the same way that it does on a C1 because the DisplayDxe.c file is setting up a 32 bit framebuffer by default.  The only difference between 24 and 32 bits is the presence of an alpha (transparency) channel, and when I boot into Ubuntu with a 32 bit framebuffer, the alpha is seen as a reserved (fixed) channel, meaning it is not actually doing anything and Ubuntu is loading what is functionally a 24 bit framebuffer.   If someone with a C1 Pi 5 knows how to check whether a true alpha channel exists in Windows or not, I know that Windows will load in either 24 or 32 bits mode.  So if it’s “true” 32 bit, I think it’s worth troubleshooting why edk2 framebuffer or GOP are failing while in the bootloader so that we can get a screen back in edk2 (RPI5-UEFI).  If we find out that Windows in C1 (the original, currently working Pi 5) doesn’t actually have a functional alpha channel, then it may as well be set to 24 bits.  Since we’re already getting an image at 24 bit setting, I think it would only take tweaking a few bits of code in DisplayDxe.c to tell it to expect and set up for 24 bits instead of 32 bits.   As far as the base problem, I did find code in the linux graphics codebase that show differences in how the alpha channel works between the two models on the software side.  However, the graphics drivers are written for linux and those changes can only help hint at what differences exist between the two products, since we are just using the basic framebuffer for EFI and don’t have access to the true linux graphics.

As it stands now, with the altered and re-compiled Pinctrl files, things boot and run fine other than graphics, which then have to be initialized by your OS, but edk2 DOES properly set up the framebuffer itself, even at 32 bit depth.   I’m on a Compute Module 5, not a Pi 5 so I’m using D0 but don’t have access to a D0 Pi 5.   That means that the changes I mentioned above would likely lead to better compatibility for other people that try those changes, since I would still need to make additional changes to the software for CM5 because it uses eMMC for storage instead of SD cards.   To boot and set up edk2 you would either need to use a Pi debug probe or other UART connector and set up edk2 from another computer or boot into it with 24 bits mode, save all your settings, switch back to 32 bits mode and then allow the bootloader to have a black screen until your OS takes control of the framebuffer and the image appears.

NumberOneGit, 2025-05-04:
I will.  Mario’s suggested method of cloning the repo using “--recurse-submodules” may have worked at the time it was created, but I had a lot of issues with it.  I think a few of the files were no longer open source, so before I will make a public fork I would need to figure out how to instruct people to clone the repo without using the convoluted workarounds I used.

NumberOneGit, 2025-05-04:
I will. Mario’s suggested method of cloning the repo using “--recurse-submodules” may have worked at the time it was created, but I had a lot of issues with it when I first cloned it in December. I think a few of the files were no longer open source, so before I will make a public fork I would need to figure out how to instruct people to clone the repo without using the convoluted workarounds I used.

Marcinoo97, 2025-05-06:
@NumberOneGit have u tested the code u committed already to your fork?

NumberOneGit, 2025-05-06:
I have been running dual boot Windows and Ubuntu since mid-February with those changes.  Framebuffer is not altered in the fork yet.  I’ve had some successes with DisplayDxe.c edits, but definitely not anything that I can post yet.   

My edits are four files in “edk2-platforms”, but I had to move many files in “edk2” from the “master” branch to RPI5-UEFI which uses the “sdmmc-dev” branch to be able to compile. 
I think as a starting point, it would be best to figure out what is unique to that branch and separate it so that the rest of the edk2 submodule can stay on the “master” branch and not continuously break as edk2 dependencies and URLs update.  At the time I started working on this in January, the original repo wasn’t archived and I didn’t plan to create a fork, just to share my findings on that repo so I didn’t document the changes I had to make in edk2 to start compiling.

For now use:
Git clone -b edk2-master-test https://github.com/NumberOneGit/rpi5-uefi

The only difference in that branch is that it corrects the “subhook” URL.

To actually compile, I think I copied CryptoPkg and UnitTestFrameworkPkg from the “master” branch of the edk2 repo to the one in RPI5-UEFI and possibly sub-folders needed to be copied over in both MdePkg and MdeModulePkg.

NumberOneGit, 2025-05-06:
I've been using the code since mid-February to boot my CM5 D0.   The four files that I altered are three related to pinctrl and one that updates board revisions to include CM5, Pi 500 and CM5 Lite models.

At the moment, I'm having a hard time reconciling the current master branch of edk2 and the out of date branch of edk2 within the original RPI5-UEFI repo in a way that compiles.   If you know of a fork that exists for C1 that has updated URLs (subhook for example, has changed from an external URL to a tianocore URL in the current edk2 master) and successfully compiles, it would be much easier for me to fork that and then add my own fork for a D0 version and additional updates.  

It's not too hard to clone both the RPI5-UEFI repo and the EDK2 repo and copy over missing dependencies from EDK2 to RPI5-UEFI, but trying to put the repo back together into an updated state that compiles and is also not publicly hosted locally may be beyond my level of patience and definitely my current level of Github experience.   I think that with the dt-bindings for the RP1 coming out soon, having a repo for community contribution would be great, but a good first step would be to work together to make a new public branch of the original repo that can be compiled as is and forked from.

Marcinoo97, 2025-05-07:
I stayed on old edk2. Idk if it compiles https://github.com/Marcinoo97/rpi5-uefi/actions/runs/14885891367 lol
Community seems dead tbh so idk

NumberOneGit, 2025-05-07:
I will post the compiled .fd file later today.  If you have any interest in trying to help figure out patches to make it compile properly, I think that is our only hope of making progress with new features.  I know that I’m way more interested in working with the actual codebase changes to make things that were broken work than trying to fix a bunch of broken dependencies because I usually work on files locally and haven’t done much on Github.  If I had the capability to host my version of edk2 submodule, I would just do that.   I opened the Issues section on my fork if you want to move the conversation over there.  

@Botspot Do you have any interest in the project?  I know that you have BVM.  I do feel like having a native working EDK2 will generate more interest from people using EXSi, FreeBSD, etc and will promote better development of ACPI tables for Pi 5.

NumberOneGit, 2025-05-07:
Okay, it’s up now.  I don’t have a D0 so please let me know if the .dtb I put in the folder is the one that is currently being used by Pi 5 users that own a D0.   I have a Compute Module 5, so it’s pretty clear which .dtb to use, but the Pi 5 .dtbs other than the original one are a bit unclear.  Or does the Pi 5 share the same .dtb between both C1 and D0 versions?
I added the one with d0 in it’s name but now I’m wondering if that was just an experimental or transitional one?

Marcinoo97, 2025-05-08:
tested 1 display and looks like its fixed.

NumberOneGit, 2025-05-08:
sounds good, i uploaded a new one just now that may be slightly faster.

PabloIvnGl, 2025-05-08:
> its probably fine, i think it was just hdmi cable problem. lets just say that its good but we need more testers just in case. 

I Really glad if you let me test the new edk2 file to try win on D model. If I understood good that's was fork compiling.

NumberOneGit, 2025-05-08:
Yes, if you want to compile it it's a little more complicated at the moment.  If you want to join the effort to fix it so it can compile properly, let me know.    The two options for compiling are:    
1.  clone both RPI5-UEFI and the current master branch of the edk2 repo and move the minimal required files from edk2 into the edk2 submodule of RPI5-UEFI local repo.
2. try to update the RPI5-UEFI files so that it will compile with the current edk2 master plus the few SD related patches Mario made to the original edk2 repo.
I prefer the second option and would definitely appreciate help getting that to work so that it can stay up to date, but I can confirm that the first option works and that's how I compiled the version that you can download on my repo.    Probably best to move all future messages over to my repo though, so we don't gum up Botspot's page with new messages.

Marcinoo97, 2025-05-08:
> Probably best to move all future messages over to my repo though, so we don't gum up Botspot's page with new messages.

but u ignore me there.

NumberOneGit, 2025-05-08:
Sorry, that comment was for Pablo, I just replied to your comment @Marcinoo97

PabloIvnGl, 2025-05-08:
I'm tryin to compile, moving files from edk2 mario's repo. to your zip. master. My cuestion is now, with the files overwriten, has to compile using ./build.sh, and what i should spect?
 config.txt
RPI_EFI.fd
BCMrpi5b.dbt
and 3 folders with boot, efi & sources. to copy in fat32partition.
really sorry my poor english

PabloIvnGl, 2025-05-08:
> I'm tryin to compile, moving files from edk2 mario's repo. to your zip. master. My cuestion is now, with the files overwriten, has to compile using ./build.sh, and what i should spect? config.txt [...]

also have to copy the edk2 original repo

NumberOneGit, 2025-05-08:
@PabloIvnGl https://github.com/NumberOneGit/rpi5-uefi/issues/1

Respond to Issue #1 in the main Issue on the new RPI5-UEFI fork and I will reply there